analogdevicesinc/linux #17: [AD9361] Issues with FIR parsing

Tey, 2016-04-28:
The second and third examples of [the documentation](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#load_a_filter) suggest it is possible to load a filter only for Rx or Tx paths. However, using the third example (Tx only) to feed `filter_fir_config` produces an error because `ad9361_parse_fir()` expects that both `tx` and `rx` variable are valid (code between lines 4963 and 4985). There's also a typo at line 4899, and a possible buffer overflow if the number of coefficients given in the file is bigger than 128.

I don't know whether this is a bug in the driver code or a documentation error, but in case of the former, a patch like [this one](https://gist.github.com/Tey/a733bb60b8bf58498312685dcca21347#file-ad9361_firsingle-diff) should be enough to fix it.

mhennerich, 2016-04-29:
Hi,

Your right - patch looks fine and applies without issues.
Can you give me your signed-off-by?

-Michael

Tey, 2016-04-29:
If nicknames are okay, then you can use that one:
`Signed-off-by: Teyut <teyut@free.fr>`

Otherwise, this is a problem as I don't want this nickname to be linked to my real name :( If it's just a matter of copyrights, I give away any copyrights I might have on that patch.

mhennerich, 2016-05-02:
Applied with this commit:

https://github.com/analogdevicesinc/linux/commit/7d47a4d1b8943128995c3a1d90b2761e5abb4c7f

Thanks for reporting and patch.

-Michael

closed 2016-05-02 by mhennerich