analogdevicesinc/linux #19: [AD9361] "adi,fagc-lock-level" property is ignored

brouvio, 2017-01-04:
Setting the DT property `adi,fagc-lock-level` in the DTSI file has no effect. This is supposed to change the [D6:D0] bits of `REG_AGC_LOCK_LEVEL` (0x101) when using fast AGC mode. However, the value of these bits is always set to the value of the `adi,agc-inner-thresh-high` property, [regardless of the AGC mode](https://github.com/analogdevicesinc/linux/blob/147070693b80c926e939c3063e9da3398109a832/drivers/iio/adc/ad9361.c#L3192) (even though this property is only valid for slow AGC mode). The value of the `adi,fagc-lock-level` property is [correctly read from the DT](https://github.com/analogdevicesinc/linux/blob/147070693b80c926e939c3063e9da3398109a832/drivers/iio/adc/ad9361.c#L8272), but never used.

The current workaround is to use `adi,agc-inner-thresh-high` as `adi,fagc-lock-level` in fast AGC mode, but this is misleading.

rgetz, 2017-01-04:
Bruno:

Thanks for reporting - the person who wrote that driver is out until Jan 9th - we will have a look when he gets back.

-Robin

brouvio, 2017-01-05:
No problem, there's no hurry as a workaround exists.

mhennerich, 2017-01-09:
Hi,

The original idea was that we can have different lock levels for fast and slow AGC. However it never got implemented since it all goes into the same register bits. We allow different AGC modes for the two RX.
So there is a conflict, since there is only one AGC configuration shared for the two RX.

I'll remove the fagc-lock-level  property to avoid confusion.

-Michael

brouvio, 2017-01-10:
Thanks for the explanations. Indeed, removing the property from source code, doc and DTSI files would be better.

mhennerich, 2018-03-16:
This property was removed - close this now.

closed 2018-03-16 by mhennerich