analogdevicesinc/linux #21: adv7511  driver  does't  work

HackLinux, 2017-02-02:
dear!
I   use  this  kernel and  adv7511  HDMI  ,but  it  doesn't work   Iadd  priintk  ,it  print   ret   <0  command can't  write  ,sometime it write  successfully,but  crt  can'display   !

static const struct reg_sequence adv7511_fixed_registers[] = {
	{ 0x98, 0x03 },
	{ 0x9a, 0xe0 },
	{ 0x9c, 0x30 },
	{ 0x9d, 0x61 },
	{ 0xa2, 0xa4 },
	{ 0xa3, 0xa4 },
	{ 0xe0, 0xd0 },
	{ 0xf9, 0x00 },
	{ 0x55, 0x02 },
};

/drivers/gpu/drm/i2c/adv7511_core.c
ret = regmap_register_patch(adv7511->regmap, adv7511_fixed_registers,
				    ARRAY_SIZE(adv7511_fixed_registers));
	if (ret)
		return ret;

lclausen-adi, 2017-02-02:
Hi,

Thanks for your request. For support please use https://ez.analog.com/community/linux-device-drivers/linux-software-drivers

- Lars

commodo, 2018-03-05:
@HackLinux @lclausen-adi 
Seems like this has been resolved.

If no objections, will close this issue tomorrow.
If I'm wrong feel free to contradict me, or re-open this.

commodo, 2018-03-06:
no objections;
closing
feel free to re-open if necessary

closed 2018-03-06 by commodo