analogdevicesinc/linux #65: altera_4.9 kernel does not enable core 1

kalden2, 2018-03-08:
altera_4.9 branch running on an Arria10 only enables core 0.

kernel config has:
CONFIG_SMP=y
CONFIG_NR_CPUS=2

[dmesg.txt](https://github.com/analogdevicesinc/linux/files/1793589/dmesg.txt)

commodo, 2018-03-08:
hey

thanks for the report;
do you happen to know/have the u-boot args ?

Alex

kalden2, 2018-03-08:
console=ttyS0,115200 mem=1536M root=/dev/mmcblk0p3 rw rootwait

kalden2, 2018-03-09:
looks like we were missing some dts changes: https://github.com/analogdevicesinc/linux/commit/ebbce1bbc4f25c0ca68f66df54ea5e8eefa90da5

This doesn't fix our problem but gets us closer.  I'm guessing the problem is on our side

kalden2, 2018-03-12:
No problem.  Looking deeper we see that secondary_start_kernel in smp.c is never getting called.  This is what alerts wait_for_completion_timeout in __cpu_up that the processor actually started up.

Attached are our dts files.
[dts.zip](https://github.com/analogdevicesinc/linux/files/1802879/dts.zip)

kalden2, 2018-03-12:
Also, here is our kernel config
[defconfig.txt](https://github.com/analogdevicesinc/linux/files/1802896/defconfig.txt)

lclausen-adi, 2018-03-12:
I believe for Arria**10** the enable-method needs to be "altr,socfpga-a10-smp".

kalden2, 2018-03-12:
yeah, we have tried both "altr,socfpga-a10-smp" and "altr,socfpga-smp".

They both result in the following dmesg output:

> Calibrating delay loop (skipped), value calculated using timer frequency.. 200.00 BogoMIPS (lpj=1000000)
> pid_max: default: 32768 minimum: 301
> Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
> Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
> CPU: Testing write buffer coherency: ok
> ftrace: allocating 22472 entries in 66 pages
> CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
> Setting up static identity map for 0x100000 - 0x100058
> CPU1: failed to come online
> Brought up 1 CPUs
> SMP: Total of 1 processors activated (200.00 BogoMIPS).
> CPU: All CPU(s) started in SVC mode.

kalden2, 2018-03-13:
alright.  we had inadvertently disabled a timer while chasing this.  with the enable-method change both cores come up now.  The BogoMIPS seems really low though (now at 400 with two cores).

rgetz, 2018-03-13:
Bogomips scales with single core CPU MHz, not number of cores.

rgetz, 2018-03-13:
But I looks like you are telling it to do this. 

Your dmesg:
Calibrating delay loop (skipped)

Something is telling the system to skip this. 

For example:
https://elinux.org/Preset_LPJ

That is where to dig.

lclausen-adi, 2018-03-14:
@rgetz on arm it also prints the some of all processors combined: https://github.com/analogdevicesinc/linux/blob/master/arch/arm/kernel/smp.c#L415-L430

lclausen-adi, 2018-03-14:
@kalden2 The reason why the bogomips is not calculated is because it doesn't have too. In this case the kernel uses a timer based delay which is more precise than the bogomips loop based one.

> Jan  1 00:00:04  kernel: [    0.000018] Switching to timer-based delay loop, resolution 10ns

kalden2, 2018-03-14:
thanks everyone.  sounds like bogomips doesnt matter and everything is fine.  so, not an issue with the 4.9 baseline at all :)

closed 2018-03-14 by kalden2