analogdevicesinc/linux #111: Windows cloning of this repo fails to checkout master.

schegde, 2018-06-08:
I was trying to clone this repository in windows, using git bash, and the checkout of master fails, with the following 2 files not found error: 
drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c
drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.h
 Turns out windows doesn't allow file/folder name creation with "aux" since it is a reserved keyword, and an underscore is prefixed when these files are created on filesystem. ("_aux.c" or "_aux.h"). Sorry if I am missing it out, but is this repository supported to work with only on a linux station? For my application, I am only interested in using some driver files (AD9694 transport layer support), and so was working with a windows dev environment.

commodo, 2018-06-11:
Hey,

Those files are `aux.c, aux.h`  are also present in upstream Linux.
( See: https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c  & https://github.com/torvalds/linux/blob/master/drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.h )
So, it's not an issue with this Linux repo in particular.
I'm surprised that Windows doesn't allow `aux` filenames; I may need to check this on a Windows system.

Which Windows version are you using ? (Win7, WIn10 ?)
Which git are you using ? The one that comes with SourceTree ? or this https://git-scm.com/ ? or cygwin's ?

To answer your question: the Linux kernel is usually built on Linux machines [sometimes Unix at best]. Building it on Windows is possible, but difficult [ requires special tooling setup ].
Cloning the Linux repo: I am not entirely sure if that works well; because there may be some filesystem differences between Linux & Windows file systems.
There could be some ways to work around those differences.

But, if all you need is the AD9694 files, you could just download them from Github directly.
Just go to this branch: https://github.com/analogdevicesinc/linux/tree/ad9694  and either download each file individually, or if you click on the `Clone or download` (green button) you can click Download zip and you get all the files for that branch in zip file (`linux-ad9694.zip`).
This (Download Zip) method is a bit faster than cloning the repo, because it won't download the git history.

schegde, 2018-06-12:
Hi,  I am using Win10, latest version of git downloaded from the git website. I tried cloning through both bash and sourcetree but same result, checkout fails because of errors in these 2 file names, and the entire version-control ends up in some weird state, showing a ton of uncommited changes on top of current HEAD, while all I did is a clone attempt. You are right, thats how I am doing it, just download the zip(Interestingly even that extraction prefixes these 2 files with '_' , but that isnt an issue, as atleast my files of interest are all in there.) But technically, I would have liked git, since I am co-ordinating with an ADI software team working on dev_ad9694 branch, and would be good to get their updates systematically as git commit pulls. Also as you pointed, I would build this kernel on a Linux station, but I would be using these driver files in my windows-heavy dev environment, which is why all these windows cloning attempt :). Thanks anyway!

commodo, 2018-06-12:
Well, if you really want, you can use Linux subsystem on Windows 10 [not available on any Windows before Win10].
It's decent, but I highly recommend you have an SSD on your system.
Link: https://docs.microsoft.com/en-us/windows/wsl/install-win10
There are a few more guides around the web on how to use it.

But that should work a bit better than using the Windows filesystem.

A lot of people also just run a Linux [usually Ubuntu] VM on a Windows box.

mhennerich, 2018-06-12:
Also if you have a Linux box. Clone the git repository there and then use [SAMBA](https://en.wikipedia.org/wiki/Samba_(software) ) and map your exported share on your windows PC.

schegde, 2018-06-14:
Thanks guys for the pointers.

closed 2018-06-14 by schegde

niuzb, 2021-07-20:
I find that it can success to  checkout the code by the WSL unbuntu(running on win10)

SkybuckFlying, 2022-02-28:
This problem is confirmed, see it for yourself in this video on youtube, where I try to checkout the linux kernel for "inspection" purposes. I would like to just view the source code to see how Linux was created. Especially before the compiler upgrade to C11. Would like to look at C89 source codes ! =D

Link to youtube video that shows the problem at timestamp 8 minutes and 20 seconds:

https://youtu.be/n15JnFgaxKs

Is there any possibility to get the checkout working ? I will also try asking this on some special git newsgroup.

welkinspring, 2022-04-07:
works on win10 git bash:
git config core.protectNTFS false
git reset --hard HEAD

gaojs, 2022-06-07:
> works on win10 git bash: git config core.protectNTFS false git reset --hard HEAD

thx!

MichaIng, 2026-02-04:
Does someone know the background why the `drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c` path can cause issues with NTFS? The [Git docs](https://git-scm.com/docs/git-config#Documentation/git-config.txt-coreprotectNTFS) mention [8.3 filenames](https://en.wikipedia.org/wiki/8.3_filename) as example, but none of the path elements is larger than 8(+3) characters, and neither is my Windows 11 host 8.3 limited, nor are longer names in any other repository an issue.

Kinda weird that it is not possible to clone Linux sources on Windows hosts without altering the Git config/flags 🤔.

mhennerich, 2026-02-04:
Just google for this _drivers/gpu/drm/nouveau/nvkm/subdev/i2c/aux.c_ - and you'll find some answers 
https://stackoverflow.com/questions/24898680/fatal-unable-to-checkout-working-tree

-Michael