analogdevicesinc/linux #143: Bugs in iio_kfifo_remove_from

allsey87, 2018-07-05:
I believe something might be something wrong with these checks in [kfifo's remove from](https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/buffer/kfifo_buf.c#L142) function. Referring to the following code, I understand that we are doing two checks. (1) we check if the fifo has enough data, i.e. enough data to remove a single sample from the buffer, and (2) we check if we were successfully able to remove that data from the fifo.

```c
if (kfifo_size(&kf->kf) < r->bytes_per_datum)
	return -EBUSY;

ret = kfifo_out(&kf->kf, data, r->bytes_per_datum);
if (ret != r->bytes_per_datum)
	return -EBUSY;
```

Unless I am missing something here, the above code just seems wrong for the following reasons:
1. `kfifo_size` returns the capacity of the buffer, not how much data is actually there, perhaps this should be `kfifo_len`?
2. `kfifo_size` returns the number of elements, not the number of bytes, so it doesn't make sense to compare the returned value to `bytes_per_datum`
3. `kfifo_out` takes the number of elements, and returns the number of elements, again the units are in elements, not bytes...

allsey87, 2018-07-05:
Adjusting the code to the following compiles and works as I would expect it to:

```c
if (kfifo_len(&kf->kf) < 1)
	return -EBUSY;

ret = kfifo_out(&kf->kf, data, 1);
if (ret != 1)
	return -EBUSY;
```

lclausen-adi, 2018-07-05:
Hi,

You are right. I believe this broke with the following change https://github.com/analogdevicesinc/linux/commit/c559afbfb08c7eac215ba417251225d3a8e01062

Want to send a patch to fix the issue?

- Lars

allsey87, 2018-07-05:
Patched attached, for some reason Github doesn't like the extension ".patch". Seems like Microsoft has gotten straight to work with leaving their mark on Github...

[0001-iio-Fix-buffer-length-checks-in-iio_kfifo_remove_fro.txt](https://github.com/analogdevicesinc/linux/files/2167188/0001-iio-Fix-buffer-length-checks-in-iio_kfifo_remove_fro.txt)

commodo, 2018-07-06:
Hey,

The patch is missing a `Signed-off-by` tag, and a description/comment of the issue in the commit comment section.
Typically for Github, we prefer pull-requests to our repo, but we can also take other forms [like patch files].

We can do a PR internally with your patch, and adapt it, referencing this discussion.
We will leave this up to you.

But, I'd propose that [if you don't [want to] do a PR], we'll do one next week sometime internally.

I say we re-open this issue until we merge this into master.
Otherwise, there are chances that we forget this.

Thanks
Alex

allsey87, 2018-07-06:
@commodo see #144. By the way, is there a road map for getting the IIO output buffer infrastructure into the mainline?

commodo, 2018-07-06:
Thanks for the PR.

So, roadmap-wise, there's always the desire/tendency to try to get most stuff in this repo into mainline.
Time-wise, it's more of a synchronicity between when we have time to push stuff and when upstream has time to look into it, then potentially repeat the review cycle for each.

FWIW, we did manage to get 2 rebased branches of our master here:
https://github.com/analogdevicesinc/linux/tree/adi-4.9.0  - should be in-sync with master ; but, to get a zero-diff we compare it with https://github.com/analogdevicesinc/linux/tree/master-sync-with-rebase-4.9 
https://github.com/analogdevicesinc/linux/tree/adi-4.14.0 - needs a bit more work;

After we update with Xilinx's master, we'll sync only `adi-4.14.0`.
The `adi-4.14.0` will be a reasonable starting point for comparing what **was** upstreamed vs what **is** already upstreamed.
In time, it should make upstreaming, comparing our master to upstream, and cherry-picking non-upstreamed patches a bit easier.

I won't give a timeline, but let's hope for an as-soon-as-possible time :)

commodo, 2018-07-06:
Closed via #144

closed 2018-07-06 by commodo