analogdevicesinc/linux #343: ADRV9009 driver errors

tfcollins, 2019-03-29:
Whenever reseting the transceiver I always see these errors:
```
adrv9009 spi0.1: ADIHAL_resetHw at index
adrv9009 spi0.1: ADIHAL_resetHw at index
adrv9009 spi0.1: ERROR: 44: TALISE_gpIntHandler(): PA Protection Channel2 Reports Error
adrv9009 spi0.1: AUX PLL lock detect reset
adrv9009 spi0.1: RF PLL lock detect reset
adrv9009 spi0.1: GP Interrupt Status 0xC0 Action: ERR_REDUCE_TXSAMPLE_PWR
adrv9009 spi0.1: ERROR: 463: TALISE_setFhmConfig: FHM range maximum frequency should be greater than minimum frequency
```
Can we handle this better?

mhennerich, 2019-05-24:
The issue is that the interrupt must be disabled before changing a profile

mhennerich, 2019-06-11:
should be fixed now

commodo, 2019-07-18:
@tfcollins @mhennerich 
can we close this?

mhennerich, 2019-07-18:
from my side - yes

commodo, 2019-07-18:
let's close it then;
we can re-open if anything is still unresolved

closed 2019-07-18 by commodo