analogdevicesinc/linux #345: [cf_axi_tdd] Typo in the cf_axi_tdd driver

tg-m, 2019-04-04:
There's a typo in the cf_axi_tdd driver:

```
drivers/iio/adc/cf_axi_tdd.c:354:  static IIO_CONST_ATTR(enable_mode_available, "rx_tx rx_only tx_onlx");
drivers/iio/adc/cf_axi_tdd.c:361:  static IIO_CONST_ATTR(dma_gateing_mode_available, "none rx_only tx_onlx rx_tx");
```

Should be `tx_only`, while it is `tx_onlx`.

dbogdan, 2019-04-04:
Thanks for reporting this. A pull request has been already made: https://github.com/analogdevicesinc/linux/pull/346

commodo, 2019-04-05:
Closed via #346

closed 2019-04-05 by commodo

tg-m, 2019-04-05:
Well... I expected something more. For example, creating `char const* const TX_ONLY_STR = "tx_only";` and changing all occurrences of this string in the source to such a constant. Similarly for `"rx_only"` and `"rx_tx"`. That would be nice. Otherwise, similar typos may occur in the future.

dbogdan, 2019-04-05:
We are open for suggestions and patches. Anyone can create pull requests - we are happy to review them. Can you do it for the mentioned changes?