analogdevicesinc/linux #360: AD9361 Register 0x00E Temperature slope 1.16 or 1.14?

zhangseven, 2019-04-22:
For AD9361 register 0x00E - Temperature

The manual says:
> The temperature word is proportional to internal die temperature with a slope of **1.16** × temperature.

But in code, ad9361_get_temp() uses **1.14** as the divider:
`return DIV_ROUND_CLOSEST(val * 1000000, 1140);`

1.16 vs 1.14,  which slope is correct?

dbogdan, 2019-04-22:
I'm pretty sure that the right coefficient is 1.16 as the latest datasheet states.
Thanks,
Dragos

zhangseven, 2019-04-23:
Thanks for the quick reply. 

I found another conflicting point:
Not like it's 1.16 in UG-671, the coefficient is 1.14 in UG-570 AD9361 Reference Manual.
And Figure 57 shows the slope is closer to 1.14 than 1.16, because the AuxADC Code at 80 C is around 156.8 rather than 159.2.

dbogdan, 2019-04-23:
We'll find out the correct value and we'll let you know about it.
Thanks,
Dragos

mhennerich, 2019-04-23:
There is another document here:
https://ez.analog.com/wide-band-rf-transceivers/design-support/w/documents/10059/ad936x-temperature-sensor

commodo, 2019-04-24:
Correct value is 1.14 .

Documentation will be updated to reflect that.

closed 2019-04-24 by commodo