analogdevicesinc/linux #368: MYKONOS_waitArmCmdStatus is blocking call

am195, 2019-05-06:
Based on the headless.c code the MYKONOS_waitInitCals can easily trigger watchdogs because of the up to 60 seconds it can be in that loop.
`  do
    {
        retVal = MYKONOS_readArmCmdStatusByte(device, opCode, cmdStatByte);
        if (retVal != MYKONOS_ERR_OK)
        {
            return retVal;
        }

        /* If error flag is non zero in [3:1], - return error */
        if ((*cmdStatByte & 0x0E) > 0)
        {
            CMB_writeToLog(ADIHAL_LOG_ERROR, device->spiSettings->chipSelectIndex, MYKONOS_ERR_ARMCMDSTATUS_ARMERROR,
                    getMykonosErrorMessage(MYKONOS_ERR_ARMCMDSTATUS_ARMERROR));
            return MYKONOS_ERR_ARMCMDSTATUS_ARMERROR;
        }

        if (CMB_hasTimeoutExpired())
        {
            return MYKONOS_ERR_WAITARMCMDSTATUS_TIMEOUT;
        }
    } while (*cmdStatByte & 0x01);`
Only looking a resolution of 1 ms there does not seem to be much point of checking so frequently. CMB_wait_ms(1) should probably be added to this loop

mhennerich, 2019-05-06:
Good point this loop is lacking the sleep element.
Thanks for pointing out.

commodo, 2019-05-08:
Fixed via #369

closed 2019-05-08 by commodo