analogdevicesinc/linux #380: Need device trees for ADRV9008-X for zynq

tfcollins, 2019-05-16:


mhennerich, 2019-05-17:
Looking for something like this?
Or do you need it for ZC706?

1. https://github.com/analogdevicesinc/linux/blob/master/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-rev10-adrv9008-1.dts
2. https://github.com/analogdevicesinc/linux/blob/master/arch/arm64/boot/dts/xilinx/zynqmp-zcu102-rev10-adrv9008-2.dts

tfcollins, 2019-05-18:
ZC706, got a request here: https://ez.analog.com/wide-band-rf-transceivers/design-support-adrv9008-1-adrv9008-2-adrv9009/f/q-a/111246/adrv9008-2-zc706-matlab-iq-stream/332189#332189

mhennerich, 2019-05-20:
Hi Dragos, (@dbogdan) 

Can you have someone porting these over to zc706?

dbogdan, 2019-05-20:
@mirceacaprioru is taking care of this.

mhennerich, 2019-05-24:
Closed via 
https://github.com/analogdevicesinc/linux/commit/5afc51f89687e0433f3990221912fefc8b1a6126

closed 2019-05-24 by mhennerich

tfcollins, 2019-05-24:
Thanks! EZ has been updated