analogdevicesinc/linux #418: Move clk/clk-adf4360.c to iio/frequency/adf4360.c

ekigwana, 2019-06-13:
Move clk/clk-adf4360.c to iio/frequency/adf4360.c to make it easier to use from userland. I propose the following tasks which each may be several steps to accomplish this.

- [ ] Move clk/clk-adf4360.c to iio/frequency/adf4360.c

- [ ] Convert into an IIO driver retaining clk-provider capability

- [ ] Add lock detect via GPIO

- [ ] Add regulator support

commodo, 2019-06-14:
Very nice task-list :)

Thanks
Alex

ekigwana, 2019-06-17:
Ran it into inflexibility of the Regmap API. When dealing with 8 bits, some can be assigned to the reg_bits and rest to val_bits. I could'nt find a way to do reg_bits 2, and val_bits 22. Also default write fn writes register first then data.

ekigwana, 2019-06-24:
@stefpopa,

This is where I am keeping a list of what I am planning to do. I'd appreciate some feedback so I don't waste anyone's time.

stefpopa, 2019-06-24:
The list looks good! :wink:

ekigwana, 2019-08-09:
Thanks all for all your efforts. Your professionalism has been outstanding! I'll start working through the process to upstream this driver.

closed 2019-08-09 by ekigwana

commodo, 2019-08-09:
Many thanks for your contributions as well :)

Since we have a direct interest in getting upstreamed as well to IIO, we would like to also help with this [since we have some experience with getting things upstreamed].

I've created a branch which is a copy of IIO.
https://github.com/analogdevicesinc/linux/tree/iio-upstream
It's a copy from here https://git.kernel.org/pub/scm/linux/kernel/git/jic23/iio.git/   branch togreg

Feel free to open PRs here and we will review them, and try to point things out, and tell you how to structure things.

Thing is, Jonathan Cameron [on the IIO subsystem] usually reviews drivers during weekends, so if you send drivers on Monday-Friday [of any week], you'll probably get an answer on Sunday the latest [assuming he's not on vacation].

This method [that I'm proposing] is to speed things up, using our experience with sending drivers to IIO, and we sometimes do it internally.
We could probably get to review our stuff in 1-2 working days.
I will be out in the last 2 weeks of August [it is vacation season], so hopefully other colleagues could take a look.

Obviously, we won't merge any of these PRs [to branch iio-upstream].
It's simply a method to use Github's review mechanism.

Patches get sent with   `git send-email <patchfile1> <patchfile2> ... <patchfileN> --to=linux-iio@vger.kernel.org --to=devicetree@vger.kernel.org --cc=robh+dt@kernel.org --cc=jic23@kernel.org`

The device-tree mailing list is needed only if changing anything related to device-trees [or adding new DTs].

ekigwana, 2019-08-09:
Thanks, I'll start working it.