analogdevicesinc/linux #503: AD9361 rates wrong

tfcollins, 2019-07-27:
The driver is producing some wrong rates when not using the FIR.

For example, setting a rate of 6 MSPS I get:

dev 'ad9361-phy', attr 'rx_path_rates', value :'**BBPLL:768000001 ADC:48000000 R2:24000000 R1:12000000 RF:6000000 RXSAMP:6000000'**
dev 'ad9361-phy', attr 'tx_path_rates', value :'**BBPLL:768000001 DAC:48000000 T2:24000000 T1:12000000 TF:6000000 TXSAMP:6000000'**

This is just a 2-2-2 decimation, but you can get a 3-2-2 decimation fine.

The implementation in libad9361 does this when you force a FIR decimation of 1. This seems to work in the "copied" driver implementation in the test code here:https://github.com/analogdevicesinc/libad9361-iio/blob/master/test/gen_rates_test.c#L78
by setting phy.rx_fir_dec and phy.tx_fir_dec to 1.

Not really sure what is different in the shipping driver now.

tfcollins, 2019-07-27:
I added some test code on this branch to help evaluate the rates and show bounds:
https://github.com/analogdevicesinc/libad9361-iio/tree/test-rates

See the GenerateRatesTest test. It will give outputs like:
```
LINUX
BBPLL | RX 1152000000 | TX 1152000000 | MAX: 1430143000
RX 288000000 (4 MAX 640000000)| TX 288000000 (4 MAX 320000000) 
RX 96000000 (3 MAX 320000000)| TX 96000000 (3 MAX 320000000) 
RX 48000000 (2 MAX 245760000)| TX 48000000 (2 MAX 245760000) 
RX 24000000 (2 MAX 122880000)| TX 24000000 (2 MAX 122880000) 
RX 6000000 (4 MAX 122880000)| TX 6000000 (4 MAX 122880000) 
----------
LIBAD9361
BBPLL | RX 1152000000 | TX 1152000000 | MAX: 1430143000
RX 288000000 (4 MAX 640000000)| TX 288000000 (4 MAX 320000000)
RX 96000000 (3 MAX 320000000)| TX 96000000 (3 MAX 320000000)
RX 48000000 (2 MAX 245760000)| TX 48000000 (2 MAX 245760000)
RX 24000000 (2 MAX 122880000)| TX 24000000 (2 MAX 122880000)
RX 6000000 (4 MAX 122880000)| TX 6000000 (4 MAX 122880000)
```

mhennerich, 2019-07-27:
I guess you didn’t set the rate governor to max ist?

tfcollins, 2019-07-27:
libad9361 doesn't attempt to play with the rate gov. I didn't think the driver did this either?

mhennerich, 2019-07-29:
AD9361 Linux driver:

	case AD9361_TRX_RATE_GOV:
		if (sysfs_streq(buf, "highest_osr"))
			st->rate_governor = 0;
		else if (sysfs_streq(buf, "nominal"))
			st->rate_governor = 1;
		else
			ret = -EINVAL;
		break;

static void ad9361_init_state(struct ad9361_rf_phy *phy)
{
	struct ad9361_rf_phy_state *st = phy->state;

	st->current_table = -1;
	st->bypass_tx_fir = true;
	st->bypass_rx_fir = true;
	st->rate_governor = 1;
	st->rfdc_track_en = true;
	st->bbdc_track_en = true;
	st->quad_track_en = true;
}

So default is 1 which means ''nominal''

In your test code:
https://github.com/analogdevicesinc/libad9361-iio/blob/master/test/gen_rates_test.c#L261

rate_governor is set to 0.

AFAICT everything behaves as expected.  The driver generates exactly the same rates as the lib under the same conditions.

mhennerich, 2019-07-30:
not a BUG - closed

closed 2019-07-30 by mhennerich