analogdevicesinc/linux #689: petalinix 2018.3 linun kernal version issue

lakshmanrajam, 2019-11-24:
i am working on adrv9009 with zcu102  up  to two days before i am able to build my pealinux project for adrv9009 project i dont know what happened with git code suddenly we are getting error while building petalinux adrv9009 projects. i seen git code committed 2 dayes before  here
https://github.com/analogdevicesinc/linux

 after this committed code i am getting error in  petalinux 2018.3 project

error : ERROR: linux-xlnx-4.14-r0 do_kernel_version_sanity_check: Package Version (4.14) does not match of kernel being built (4.19). Please update the PV variable to match the kernel source or set KERNEL_VERSION_SANITY_SKIP="1" in your recipe.

can u please help me in this .??

note: i changed PV variable in   meta-adi/meta-adi-xilinx/recipes-kernel/linux/linuxxlnx_%.bbappend

but i am getting licence issue .

mhennerich, 2019-11-25:
We updated our master branch very recently to Linux Version 4.19.
It's work in progress and breakage is expected. Also petalinux workflow is not yet tested.
Until we do a release, please use our currently 2019_R1 release branch.

https://github.com/analogdevicesinc/linux/tree/2019_R1

-Michael

lakshmanrajam, 2019-11-25:
Hi,
Should we leave the branch as master itself ?
Because, when we tried with the 2019_R1 github repo, we got the following error:

 ls-remote https://github.com/analogdevicesinc/linux/tree/2019_R1  failed with exit code 128, output:
fatal: repository 'https://github.com/analogdevicesinc/linux/tree/2019_R1/' not found

Can you please help us on this ?

Thank You.

commodo, 2019-11-25:
So, the Petalinux layer we have is pretty new.
We'll make some effort to integrate it into our CI.

In the meantime, I'll do some tests, and create some release branches.
I am pretty sure, that initially this was created based on our 2018_R2 release.
So, now there is a https://github.com/analogdevicesinc/meta-adi/tree/2018_R2 branch.

I'll create a 2019_R1 branch and test it.
Link will be https://github.com/analogdevicesinc/meta-adi/tree/2019_R1

For master, we will see. We still need to define a process for Petalinux.

> ls-remote https://github.com/analogdevicesinc/linux/tree/2019_R1 failed with exit code 128, output:
> fatal: repository 'https://github.com/analogdevicesinc/linux/tree/2019_R1/' not found

Regarding this, this looks weird.
I don't know what that error means, but we do have a 2019_R1 branch on our Github repo for quite some time. Maybe you need to tell Petalinux to fetch the 2019_R1 branch somehow.

lakshmanrajam, 2019-11-25:
Thanks for the reply commodo.
It seems that, the version with 4.14 is broken now, it was working till 4-days back.
We have been using ADI linux git master in sync with petalinux 4.14 and is no more working.

But, is there a commit-id/revision that we can use in meta-adi/meta-adi-xilinx/recipes-kernel/linux/linux-xlnx-%.bbappend ?

commodo, 2019-11-25:
You can use 2019_R1 or 2018_R2 here https://github.com/analogdevicesinc/meta-adi/blob/master/meta-adi-xilinx/recipes-kernel/linux/linux-xlnx_%25.bbappend#L2

lakshmanrajam, 2019-11-25:
When we use 2018_R2 in linux-xlnx_%.bbappend as follows:

> BRANCH = "2018_R2" and KBRANCH = "master"

We got the following error at the end of the build:

> ERROR: linux-xlnx-4.14-r0 do_kernel_version_sanity_check: Package Version (4.14) does not match of kernel being built (4.19). Please update the PV variable to match the kernel source or set KERNEL_VERSION_SANITY_SKIP="1" in your recipe.
ERROR: linux-xlnx-4.14-r0 do_kernel_version_sanity_check: Function failed: do_kernel_version_sanity_check (log file is located at /home/lakshman/workspace/adrv/11ref/build/tmp/work/plnx_zynqmp-xilinx-linux/linux-xlnx/4.14-r0/temp/log.do_kernel_version_sanity_check.20838)
ERROR: Logfile of failure stored in: /home/lakshman/workspace/adrv/11ref/build/tmp/work/plnx_zynqmp-xilinx-linux/linux-xlnx/4.14-r0/temp/log.do_kernel_version_sanity_check.20838
ERROR: Task (/opt/pkg/petalinux/2018.3/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2018.3.bb:do_kernel_version_sanity_check) failed with exit code '1'

But when we keep:
> BRANCH = "2018_R2" and KBRANCH = "2018_R2", we get the following error right at the start:

>  checkout and clobber any unimportant files
	git checkout -f ${machine_branch}
 which triggered exception FetchError: Fetcher failure: Unable to resolve '2018_R2' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git

I'm totally confused on what could be the right combination between Xilinx git and ADI's. Appreciate your help.
Thank You.

alexarsenault, 2019-11-25:
Also experiencing this issue and would love to hear of some resolution.

bkzshabbaz, 2019-11-26:
I tried to change the branch from master to 2019_R1:
```
DESCRIPTION = "ADI kernel"
BRANCH = "2019_R1"
# always use latest source revision
SRCREV = "${AUTOREV}"
SRC_URI += "git://github.com/analogdevicesinc/linux.git;protocol=https;branch=${BRANCH}"
KBRANCH = "${BRANCH}"
PV = "4.14"
```
But get this error:
`FetchError: Fetcher failure: Unable to resolve '2019_R1' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git`
Why is it trying to grab from linux-xlnx?

nunojsa, 2019-11-26:
This is odd. It might be a regression due to [838bb11 ](https://github.com/analogdevicesinc/meta-adi/commit/838bb11217c41edd504ae210b2368e3847864a4d). For a quick try, you can do `SRC_URI="git://github.com/analogdevicesinc/linux.git;protocol=https;branch=${BRANCH}" ` instead of `+=`. I did recompiled before pushing this fix and it was fetching ADI sources for me but this may need a little bit more testing. Note that with this change using `petalinux-config -c kernel` won't work anymore. Alternately, before trying the above, maybe clean your kernel build with `petalinux-build -c kernel -x distclean` and try to build it again...

@lakshmanrajam you should have `BRANCH = "2018_R2" and KBRANCH = "${KBRANCH}"`. Also try the above and see if fixes the issue for you

lakshmanrajam, 2019-11-27:
HI all,
I resolved this issue  by using 2019_R1 branch 

> DESCRIPTION = "ADI kernel"
BRANCH = "2019_R1"
always use latest source revision
SRCREV = "${AUTOREV}"
SRC_URI += "git://github.com/analogdevicesinc/linux.git;protocol=https;branch=${BRANCH}"
KBRANCH = "${BRANCH}"
PV = "4.14"

but i got error 

> FetchError: Fetcher failure: Unable to resolve '2019_R1' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git

for this i did one modification in petalinux installed directory 

i changed xilinx git hub to analog git  in **linux-xlnx.inc** file

> #KERNELURI ??= "git://github.com/Xilinx/linux-xlnx.git;protocol=https"
KERNELURI ??= "git://github.com/analogdevicesinc/linux.git;protocol=https"

note: dont try this branch 2018_R2 we are unable to up the adrv9009 in zcu102 board.

Thank you.

nunojsa, 2019-11-28:
> #KERNELURI ??= "git://github.com/Xilinx/linux-xlnx.git;protocol=https"
KERNELURI ??= "git://github.com/analogdevicesinc/linux.git;protocol=https"

Yes, that works but it is clearly something we cant do in our layer (we have to fix it in our own recipe extension). Apparently, xilinx kernel recipe is doing some checks for the KBRANCH existence in all remotes. With master, everything works because master exists in both our and xilinx remotes. And, in the end, our tree is used for the build. As soon as we set a branch that does not exist in one of the remotes, things apparently break. This makes sense, because when I was testing [838bb11](https://github.com/analogdevicesinc/meta-adi/commit/838bb11217c41edd504ae210b2368e3847864a4d) I only used master. Anyways, unless we can allocate someone to fix this, I will be able to look at this only next year (most likely). Maybe setting `KERNELURI ?= "git://github.com/analogdevicesinc/linux.git;protocol=https"` or more drastically `KERNELURI = "git://github.com/analogdevicesinc/linux.git;protocol=https"` in our recipe can be enough for this. We then need to document properly how to change the kernel branch to build (changing `PV` in addition to `BRANCH` might be needed). Alternately we can have a recipe for each kernel version we want to support and change the kernel version to build in the build conf file...

commodo, 2020-01-20:
@lakshmanrajam 
there have been some updates to the meta-adi repo;
can you retry?

i'd close the issue here/now;
feel free to re-open if there are [more] issues

closed 2020-01-20 by commodo