analogdevicesinc/linux #787: petalinux FetchError: Unable to resolve '2018_R2' in upstream git repository in git ls-remote output for github.com/analogdevicesinc/libad9361-iio.git

esraltun, 2019-12-25:
I worked with petalinux v2018.3, external linux kernel (adi-4.14.0 branch) and added meta-adi layers (master branch)
I didn't solve this issue. How can I fix it?

esra@esra-MS-7A38:~/5g/petalinux_projects/xilinx-zcu102-2018.3$ petalinux-build [INFO] building project
[INFO] sourcing bitbake
INFO: bitbake petalinux-user-image
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 1373 entries from dependency cache.
ERROR: ExpansionError during parsing /home/esra/meta-adi/meta-adi-core/recipes-core/libad9361-iio/libad9361-iio_0.2.bb
Traceback (most recent call last):
  File "/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/data_smart.py", line 412, in DataSmart.expandWithRefs(s="${@bb.parse.BBHandler.vars_from_file(d.getVar('FILE', False),d)[1] or '1.0'}+git${SRCPV}", varname='PV'):
                 try:
    >                s = __expand_var_regexp__.sub(varparse.var_sub, s)
                     try:
  File "/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/data_smart.py", line 111, in VariableParse.var_sub(match=<_sre.SRE_Match object; span=(80, 88), match='${SRCPV}'>):
                 else:
    >                var = self.d.getVarFlag(key, "_content")
                 self.references.add(key)
  File "/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/data_smart.py", line 794, in DataSmart.getVarFlag(var='SRCPV', flag='_content', expand=True, noweakdefault=False, parsing=False):
                     cachename = var + "[" + flag + "]"
    >            value = self.expand(value, cachename)
     
  File "/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/data_smart.py", line 436, in DataSmart.expand(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
         def expand(self, s, varname = None):
    >        return self.expandWithRefs(s, varname).value
     
  File "/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/data_smart.py", line 426, in DataSmart.expandWithRefs(s='${@bb.fetch2.get_srcrev(d)}', varname='SRCPV'):
                 except Exception as exc:
    >                raise ExpansionError(varname, s, exc) from exc
     
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Unable to resolve '2018_R2' in upstream git repository in git ls-remote output for github.com/analogdevicesinc/libad9361-iio.git

Summary: There was 1 ERROR message shown, returning a non-zero exit code.
ERROR: Failed to build project

commodo, 2019-12-25:
Hey,

Thanks for reporting.
I've created the branch.

Can you retry?

Thanks
Alex

esraltun, 2019-12-26:
Hi,
I try but not change

esra@esra-MS-7A38:~/5g/petalinux_projects/xilinx-zcu102-2018.3$ petalinux-config -c kernel
[INFO] generating Kconfig for project
[INFO] sourcing bitbake
[INFO] generating plnxtool conf
[INFO] generating meta-plnx-generated layer
[INFO] configuring: kernel
[INFO] generating kernel configuration files
[INFO] bitbake virtual/kernel -c menuconfig
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 2072 entries from dependency cache.
ERROR: /home/esra/petalinux_20183/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2018.3.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 5, function: <module>
     0001:__anon_22__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_25__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_patch_bbclass(d)
     0003:__anon_160__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_base_bbclass(d)
     0004:__anon_686__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_base_bbclass(d)
 *** 0005:__anon_129__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_externalsrc_bbclass(d)
     0006:__anon_20__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_blacklist_bbclass(d)
     0007:__anon_1305__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_insane_bbclass(d)
     0008:__anon_250__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_package_bbclass(d)
     0009:__anon_711__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_package_rpm_bbclass(d)
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/meta/classes/externalsrc.bbclass', lineno: 64, function: __anon_129__home_esra_petalinux_20183_components_yocto_source_aarch64_layers_core_meta_classes_externalsrc_bbclass
     0060:            d.setVar('B', '${WORKDIR}/${BPN}-${PV}/')
     0061:
     0062:        local_srcuri = []
     0063:        fetch = bb.fetch2.Fetch((d.getVar('SRC_URI') or '').split(), d)
 *** 0064:        for url in fetch.urls:
     0065:            url_data = fetch.ud[url]
     0066:            parm = url_data.parm
     0067:            if (url_data.type == 'file' or
     0068:                    'type' in parm and parm['type'] == 'kmeta'):
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/__init__.py', lineno: 1590, function: __init__
     1586:
     1587:        for url in urls:
     1588:            if url not in self.ud:
     1589:                try:
 *** 1590:                    self.ud[url] = FetchData(url, d, localonly)
     1591:                except NonLocalMethod:
     1592:                    if localonly:
     1593:                        self.ud[url] = None
     1594:                        pass
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/__init__.py', lineno: 1261, function: __init__
     1257:            logger.warning('Consider updating %s recipe to use "protocol" not "proto" in SRC_URI.', d.getVar('PN'))
     1258:            self.parm["protocol"] = self.parm.get("proto", None)
     1259:
     1260:        if hasattr(self.method, "urldata_init"):
 *** 1261:            self.method.urldata_init(self, d)
     1262:
     1263:        if "localpath" in self.parm:
     1264:            # if user sets localpath for file, use it instead.
     1265:            self.localpath = self.parm["localpath"]
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/git.py', lineno: 239, function: urldata_init
     0235:        write_tarballs = d.getVar("BB_GENERATE_MIRROR_TARBALLS") or "0"
     0236:        ud.write_tarballs = write_tarballs != "0" or ud.rebaseable
     0237:        ud.write_shallow_tarballs = (d.getVar("BB_GENERATE_SHALLOW_TARBALLS") or write_tarballs) != "0"
     0238:
 *** 0239:        ud.setup_revisions(d)
     0240:
     0241:        for name in ud.names:
     0242:            # Ensure anything that doesn't look like a sha256 checksum/revision is translated into one
     0243:            if not ud.revisions[name] or len(ud.revisions[name]) != 40  or (False in [c in "abcdef0123456789" for c in ud.revisions[name]]):
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/__init__.py', lineno: 1291, function: setup_revisions
     1287:
     1288:    def setup_revisions(self, d):
     1289:        self.revisions = {}
[... 18 lines cut ...]
     1548:        try:
     1549:            return revs[key]
     1550:        except KeyError:
 *** 1551:            revs[key] = rev = self._latest_revision(ud, d, name)
     1552:            return rev
     1553:
     1554:    def sortable_revision(self, ud, d, name):
     1555:        latest_rev = self._build_revision(ud, d, name)
File: '/home/esra/petalinux_20183/components/yocto/source/aarch64/layers/core/bitbake/lib/bb/fetch2/git.py', lineno: 581, function: _latest_revision
     0577:                sha1, ref = l.split()
     0578:                if s == ref:
[... cut ...]
     0585:        Compute the latest release name like "x.y.x" in "x.y.x+gitHASH"
Exception: bb.fetch2.FetchError: Fetcher failure: Unable to resolve '2018_R2' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git

ERROR: Failed to parse recipe: /home/esra/petalinux_20183/components/yocto/source/aarch64/layers/meta-xilinx/meta-xilinx-bsp/recipes-kernel/linux/linux-xlnx_2018.3.bb

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
ERROR: bitbake failed to cleansstate kernel
ERROR: Failed to config kernel.

commodo, 2020-01-06:
> I worked with petalinux v2018.3, external linux kernel (adi-4.14.0 branch) and added meta-adi layers (master branch)

Please don't use the adi-4.14.0 branch; it's obsolete now.
Either use one of the release branches, 2018_R2, 2019_R1, etc.

I'll take a look closer at this.

esraltun, 2020-01-13:
Thank you for reply. I solved my problem.  I used default linux kernel (Linux Component Selection-->linux-kernel (linux-xlnx)) in petalinuxv2018.3 ,  analog devices referance design system.hdf (Vivado 2018.3) and meta-adi (master branch). When I built this system, it failed ( Fetcher failure: Unable to resolve '2018_R2' in upstream git repository in git ls-remote output for github.com/Xilinx/linux-xlnx.git).
 I changed BRANCH="master" and PV="4.19" into meta-adi-xilinx/recipe-kernel/linux/linux-xlnx_%.bbappend file.
Fix it.

nunojsa, 2020-01-13:
Yeps. That should fix it. But we updated our meta-adi repo recently and fixed things differently. Now, there's a dedicated 2018_R2 branch if you want to work with that release for ADI tools...

commodo, 2020-01-16:
Let's close this issue.

If it's still an issue we can re-open.

closed 2020-01-16 by commodo