analogdevicesinc/linux #788: build of iio/adc/Kconfig. This particular line results in error

vlsireddy, 2020-01-03:
Hi Team@ADI,
                     For some diag/debug purpose, i had to merge code from adi linux to xilinx linux and during the build, this particular Kconfig results in build error. 
 
Kconfig error: "recursive dependency detected"

with below lines
======================
config **AD7606**
	tristate
	select IIO_BUFFER
	select IIO_TRIGGERED_BUFFER

**config AD7606_IFACE_PARALLEL**
	tristate "Analog Devices AD7606 ADC driver with parallel interface support"
	depends on HAS_IOMEM
	**select AD7606**
	help
	  Say yes here to build parallel interface support for Analog Devices:
	  ad7605-4, ad7606, ad7606-6, ad7606-4 analog to digital converters (ADC).

	  To compile this driver as a module, choose M here: the
	  module will be called ad7606_parallel.

**config AD7606_IFACE_SPI**
	tristate "Analog Devices AD7606 ADC driver with spi interface support"
	depends on SPI
	**select AD7606**
	help
	  Say yes here to build spi interface support for Analog Devices:
	  ad7605-4, ad7606, ad7606-6, ad7606-4 analog to digital converters (ADC).

	  To compile this driver as a module, choose M here: the
	  module will be called ad7606_spi.

======================
Thanks

commodo, 2020-01-06:
When merging this change/driver, do you also still have [by any chance] the old `drivers/staging/adc/ad7606.c` files ?

If so, make sure to remove it, otherwise conflicts [with it] are expected.
That driver was moved out of staging in the ADI tree and in the upstream tree, and git history may be a bit confused. That may also mean we've probably not done some good git strategies, but we'll improve on that.

commodo, 2020-01-16:
@vlsireddy 
ping?

if no reply within a few days, i'd close this;
we can be re-open it if nedeed;

closed 2020-01-20 by commodo