analogdevicesinc/linux #822: AD9361 driver does not update recalculated rf_bandwidth value

tfcollins, 2020-01-14:
The rf_bandwidth cannot be set to any value, and the driver correctly recalculates it based on the current BBPLL and available dividers. However, this is not written back to the actual struct which is read back for the attribute. So you have no idea what the actual value is. This is true for TX and RX.

mhennerich, 2020-01-14:
The real RX and TX RFBW is a composite of multiple filters. Which filter BW calculation would you use to report?

tfcollins, 2020-01-14:
I would say once the caldiv is determined here https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/adc/ad9361.c#L2473 , use that and the BBPLL to update tx_bb_bw.

mhennerich, 2020-01-15:
Can you provide more details on how to derive the exact BW from the target using the caldiv?
For RX which value would you use?

tfcollins, 2020-01-15:
For RX:
1. Calculate caldiv with `caldiv = ceil(BBPLL * ln(2) / (BWrequested * 2.8 * pi))`
2. Get the actual bandwidth with `BWactual = BBPLL * ln(2)/(2.8*pi*caldiv)`

For TX:
1. Calculate caldiv with `caldiv = ceil(BBPLL * ln(2) / (BWrequested * 3.2 * pi))`
2. Get the actual bandwidth with `BWactual = BBPLL * ln(2)/(3.2*pi*caldiv)`