analogdevicesinc/linux #987: ad7124 probe bug

rgetz, 2020-04-28:
https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/adc/ad7124.c#L690

Doesn't probe for the device, and fail if there isn't anything attached. This means that enabling the driver in the kernel makes it show up, even if no device is actually attached.

There is a device ID register, that we do not check.

commodo, 2020-04-28:
well, this is more of a general sweep we should do over our drivers;
technically, for SPI, the only way to validate that the device is there or not, is to read the device-ID [which is usually chip-specific];
some drivers do this check, some don't;

it could be that more drivers would initialize just by being added into the device-tree;

rgetz, 2020-05-20:
Right - sine the driver is (normally) chip specific - reading chip specific registers, and checking chip specific contents should be easy.

Some chips don't have a chip ID, so this isn't possible. The AD7124 does - so for this specific driver it's easy. This should be something that is on the review checklist for all reviewers.

Since it was last touched by @mirceacaprioru  and @tachicialex - maybe who ever has hardware (to make sure it probes properly when connected, and fails gracefully when not) can have a look.

-Robin

commodo, 2020-05-21:
hmm; it might be a bit more tricky for this chip as well;
looking at the datasheets for the ID regs:
https://www.analog.com/media/en/technical-documentation/data-sheets/AD7124-4.pdf  (reset values 0x04, 0x06)
https://www.analog.com/media/en/technical-documentation/data-sheets/AD7124-8.pdf (reset values 0x14, 016)

the first nibble is the device_id (0x0 or 0x1) and the second nibble is the silicon revision 0x4 or 0x6 ;
IIRC, SPI reads 0 if there is nothing there;
and i wouldn't do an exact matching of the silicon revision, as that can change later, and updating it is a bit of a hassle [with upstreaming and everything];

maybe something of a compromise would be to do exact matching on the first nibble, and check that the silicon revision is non-zero;
that would work for both chips and be future-proof;

commodo, 2020-05-21:
i've created a patch here;
can you test it?
https://github.com/analogdevicesinc/linux/commits/ad7124-device-id

rgetz, 2020-05-22:
NP - I will test tomorrow.

commodo, 2020-07-21:
tested by @tachicialex

closed 2020-07-21 by commodo