analogdevicesinc/linux #1007: iio: ad9517: VCO divider

Liambeguin, 2020-05-11:
Hi,

Looking at `drivers/iio/frequency/ad9517.c` it seems like the driver cannot update the VCO divider at runtime.
I'm not familiar with the clock framework but would converting `st->div0123_freq` to `hw_clk` structures and making it a parent to the `ad9517_clk_div` structs be the right approach?
I imagine updating the rate of a parent clock will trigger a `set_rate` for all children.

Thanks, 
Liam

rgetz, 2020-05-20:
@Liambeguin : best place for support/questions - is ADI's EngineerZone; I posted your question:

https://ez.analog.com/linux-software-drivers/f/q-a/165706/iio-ad9517-vco-divider

Liambeguin, 2020-05-21:
Thanks, I'll follow progress there

commodo, 2020-06-03:
Will close this for now.
Since moved to EZ.

closed 2020-06-03 by commodo