analogdevicesinc/linux #1029: ad_adc and Kconfig

Liambeguin, 2020-06-03:
Hi, 

The generic [ad_adc.c](https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/adc/Makefile#L80) driver seems to not be linked to the right Kconfig option.

Should it be moved to something like this?
```diff
-obj-$(CONFIG_ADMC) += admc_adc.o admc_speed.o admc_ctrl.o ad_adc.o
+obj-$(CONFIG_ADMC) += admc_adc.o admc_speed.o admc_ctrl.o
+ 
+obj-$(CONFIG_AD_ADC) += ad_adc.o
```

Thanks,

commodo, 2020-06-04:
short answer: yes
long answer: this driver will go away at some point, in favor of this https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/tree/drivers/iio/adc/adi-axi-adc.c  ; so no idea if it's worth doing that; if there's a direct need for that, then sure, but if it can wait to be replaced by the upstreamed version, i would do it;
[when it was upstreamed, the name was suggested to make it ADI_AXI_ADC]

there's a couple of these ADC drivers that were copied around for a while, based on their direct requirement for a certain project;
the more common one actually is `cf_axi_adc`, but it's too complicated on it's own;
so, when upstreaming, i try to pick out parts needed for each driver that goes upstream with this driver (ADI_AXI_ADC);

Liambeguin, 2020-06-04:
Hi, 

Thanks for your detailed answer.
I don't really need to make the change, I just thought I'd flag it as I noticed the issue.
Thanks again,

closed 2020-06-04 by Liambeguin