analogdevicesinc/linux #1032: What is "the AD9371 API license"?

jkmarz, 2020-06-09:
Hi! 

The files in linux/drivers/iio/adc/mykonos/ are marked with the following statement: "Released under the AD9371 API license, for more information see the "LICENSE.txt" file in this zip file."

There is no LICENSE.txt file to be found, nor are the files on GitHub being presented in a zip file (I presume the statement is an artifact left over from a time when the files *were* distributed in a zip file, and accompanied by a LICENSE.txt file).

Can you please clarify what license applies to the files in linux/drivers/iio/adc/mykonos/ ?

Many thanks!

dbogdan, 2020-06-09:
Hi,
Those references to LICENSE.txt are not valid, sorry for that - the actual content of that file is: https://www.analog.com/en/license/licensing-agreement/transceiver-evaluation-software.html
We'll fix this issue.
Thanks,
Dragos

jkmarz, 2020-06-09:
Thanks a ton!

commodo, 2020-06-11:
I'm assuming this is resolved.
So, I'll close it.

Feel free to re-open if there's anything related.

closed 2020-06-11 by commodo