analogdevicesinc/linux #1060: ad5593r/ad5592r mutual exclusion violation in driver

cstanhope, 2020-07-02:
In drivers/iio/dac/ad5592r-base.c there appears to be a small violation of a mutex used in the ad5592r_read_raw() function. This issue manifested itself when attempting to read different ADC channels from different processes, which generated EIO errors somewhat randomly in any of the participating processes. Additionally, error messages in the kernel log were generated like the following: "Error while reading channel 4". This indicated that an ADC reading came from a different channel than expected. Reading from a single process never caused problems.

An example that should hopefully recreate the problem, assuming you have three consoles open and sitting in sysfs path for the device (e.g. /sys/bus/iio/devices/iio:device0), in one shell you can execute: 

for (( i = 0; i < 10000; i++)) { cat in_voltage0_raw ; } ;

And in another shell:

for ((i = 0; i < 10000; i++)) { cat in_voltage4_raw ; } ;

If just those two shells are active (assuming from a fresh boot or modprobe), things seem to be okay. But in the third shell execute:

for ((i = 0; i < 10000; i++)) { cat in_temp_scale ; } ;

This should produce many examples of the errors. This will persist even if you stop the third loop with the other two running. I believe I have tracked it down to this code path: 

https://github.com/analogdevicesinc/linux/blob/6edc6cd24b486f5f3baff016de053bfd009b762e/drivers/iio/dac/ad5592r-base.c#L416

If the condition is true, the mutex is never locked. However, the flow of control proceeds to the end of the function where the mutex is then unlocked, which isn't great for any other process holding the lock. A small patch allowed me to test the idea. With the patch applied, I have stopped receiving the EIO errors whether in my code or the above examples. 

Although the mainline kernel has switched over to using a private mutex for the device in this driver, this same code path exists there as well.
```diff
diff --git a/drivers/iio/dac/ad5592r-base.c b/drivers/iio/dac/ad5592r-base.c
index 095530c233e4..055e45b29cc5 100644
--- a/drivers/iio/dac/ad5592r-base.c
+++ b/drivers/iio/dac/ad5592r-base.c
@@ -417,7 +417,8 @@ static int ad5592r_read_raw(struct iio_dev *iio_dev,
                        s64 tmp = *val * (3767897513LL / 25LL);
                        *val = div_s64_rem(tmp, 1000000000LL, val2);
 
-                       ret = IIO_VAL_INT_PLUS_MICRO;
+                       /* We didn't take the lock and there's no more work to do. */
+                       return IIO_VAL_INT_PLUS_MICRO;
                } else {
                        int mult;
```

commodo, 2020-07-03:
good catch
do you want to send a patch, or shall we apply it ourselves?
same question for the mainline patch

as far as i can tell, this patch should apply in both trees;

cstanhope, 2020-07-05:
I'm okay with you sending the patches if you can. If you can't, I can try to make time to do so. My main concern is getting it upstream so I don't run into it again. :)

mirceacaprioru, 2020-07-06:
Hi do you have an email address to add you as Reported-by ?

cstanhope, 2020-07-06:
Sure! You can use charles.stanhope@gmail.com. Thank you!

commodo, 2020-07-07:
Fixed via #1061

closed 2020-07-07 by commodo