analogdevicesinc/linux #1274: Raw values of adxrs290 were not sign-extended.

kister-jimenez, 2020-11-12:
https://github.com/analogdevicesinc/linux/blob/2c323fd8c3c666fb27325730f44975d314274361/drivers/iio/gyro/adxrs290.c#L114

The raw value of this attribute is a 16-bit signed integer. However, the 16-bit spi data was assigned with to an int without sign-extension. This will cause the attribute to be positive and (doubled) in negative revolutions.

Same thing with the temperature attribute. Temperature is signed 12-bit stored as signed 16-bit integer but, was stored as int without sign extension. https://github.com/analogdevicesinc/linux/blob/2c323fd8c3c666fb27325730f44975d314274361/drivers/iio/gyro/adxrs290.c#L134

commodo, 2020-11-12:
@layman-n-ish 
can you take a quick look at this?

layman-n-ish, 2020-11-12:
Hi @kister-jimenez. Thanks for reporting this. 

You're right, the raw two's complement rate and temperature data from ADXRS290 should be sign-extended while packing those to an _int_. I guess I didn't notice this because I expected the readings exposed to the sysfs attribute files to be the raw, two's-complemented values.

Modifying [L127](https://github.com/analogdevicesinc/linux/blob/2c323fd8c3c666fb27325730f44975d314274361/drivers/iio/gyro/adxrs290.c#L127) to `*val = sign_extend32(temp, 15);` should fix the issue, I suppose.

Similarly, [L149](https://github.com/analogdevicesinc/linux/blob/2c323fd8c3c666fb27325730f44975d314274361/drivers/iio/gyro/adxrs290.c#L149) to `*val = sign_extend32(temp & 0x0FFF, 11);`

The buffer data capture code is not susceptible as, to my knowledge, the IIO core does the necessary extension. So, nothing changes there.

I tested the above modifications and hope this is what you expected:

[image]

Feel free to send in a PR fixing this issue once you too test the above modifications.

kister-jimenez, 2020-11-13:
@layman-n-ish Thanks. I'm closing this now. I've tested the fix and the PR has been merged.

closed 2020-11-13 by kister-jimenez