analogdevicesinc/linux #1461: Unable to restore RSSI gain step error data

ekigwana, 2021-03-03:
https://github.com/analogdevicesinc/linux/blob/5c783ae1fbdc0299d5778961917509ada10d8f4f/drivers/iio/adc/ad9361.c#L7173

looks like:
```static IIO_DEVICE_ATTR(rssi_gain_step_error, S_IRUGO,``
should be:
```static IIO_DEVICE_ATTR(rssi_gain_step_error, S_IRUGO | S_IWUSR,```

Is this something that Linux users have not needed? Or is the function not achieve anything practical? I am filtering IQ data and using 10 log 10(I**2 + q**2) - rx_hardwaregain_chan0 + offset to get absolute but seems like mag squared values have a std deviation of at least 2.5 dBm even in manual gain mode. <- I'll ask about this bit in the forum

mhennerich, 2021-03-04:
Looks like a bug -
store/write support was added with this commit https://github.com/analogdevicesinc/linux/commit/147070693b80c926e939c3063e9da3398109a832
but the permissions we're not set.
We fix this soon.

nunojsa, 2021-03-04:
Should be fixed with #1463

closed 2021-03-04 by nunojsa