analogdevicesinc/linux #1464: Ez question

mirceacaprioru, 2021-03-05:
We are getting EZ question regarding booting issues on Talise SOM. They are related to this switch should we just remove it from the device tree ? 
Here is the topic: https://ez.analog.com/linux-software-drivers/f/q-a/541904/adrv9009-zu11eg-not-booting-from-newest-image-2019_r2

https://github.com/analogdevicesinc/linux/blob/89450f163b0404eb7248b5d6ba9b4ca8cc94a675/arch/arm64/boot/dts/xilinx/zynqmp-adrv9009-zu11eg-reva-adrv2crr-fmc-reva.dts#L107

dbogdan, 2021-03-05:
Only documenting this in case it hasn't been done already, should be enough from my point of view.

mhennerich, 2021-03-05:
There is no LID on this platform, we can do ourselves and customers a favor and remove this input key/switch event.
Before the next one runs into that. And I'm sure they will ...

dbogdan, 2021-03-05:
Should we review all the keys/buttons then?

mhennerich, 2021-03-05:
Well - use something that doesn't negatively impact the device.
I think this is for testing only. So instead of SW_LID we could use  SW_LINEIN_INSERT

closed 2021-03-08 by mirceacaprioru