analogdevicesinc/linux #1494: cant build ad7768 as part of kernel

jamesoncollins, 2021-03-28:
Trying to compile ad7768 as part of the kernel results in a linker error:

```
WARNING: "return_address" [vmlinux] is a static EXPORT_SYMBOL_GPL
  MODINFO modules.builtin.modinfo
  LD      .tmp_vmlinux1
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768.o: in function `hw_submit_block':
ad7768.c:(.text+0x100): undefined reference to `iio_dmaengine_buffer_submit_block'
/home/usert/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768.o: in function `ad7768_probe':
ad7768.c:(.text+0x834): undefined reference to `devm_iio_dmaengine_buffer_alloc'
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768.o:(.rodata+0x84): undefined reference to `iio_dmaengine_buffer_abort'
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768-1.o: in function `hw_submit_block':
ad7768-1.c:(.text+0xd4): undefined reference to `iio_dmaengine_buffer_submit_block'
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768-1.o: in function `ad7768_probe':
ad7768-1.c:(.text+0xc68): undefined reference to `iio_dmaengine_buffer_alloc'
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: ad7768-1.c:(.text+0xc7c): undefined reference to `iio_dmaengine_buffer_free'
/home/user/build_test_light/host/bin/arm-buildroot-linux-gnueabihf-ld: drivers/iio/adc/ad7768-1.o:(.rodata+0xa8): undefined reference to `iio_dmaengine_buffer_abort'
Makefile:1077: recipe for target 'vmlinux' failed
make[2]: *** [vmlinux] Error 1
```

mirceacaprioru, 2021-03-29:
Hi,

What branch are you using and what configuration? It seems that some dependencies have not been enabled for the driver ?

jamesoncollins, 2021-03-29:
The branch is 2019_r2

The config file is attached.

I'm assuming this has to do with these two lines.  IIRC, these entries cannot be configured from menuconfig, they are meant to be selected by a driver.

```
CONFIG_IIO_BUFFER_DMA=m
CONFIG_IIO_BUFFER_DMAENGINE=m
```

mirceacaprioru, 2021-03-29:
Yes you are right about those configurations. The thing is that if a dependency is built as a module you can't  build a driver integrated in the kernel that has that dependency, only to enable it as a module also.

If you do not see them inside menuconfig that means that something that they depend on is not enabled. Use the search option in menuconfig to determine what a feature depends on and enable that beforehand. This will ensure you do not miss anything.

Editing things in a text editor won't guarantee you did not miss anything. 

Regards,
Mircea

jamesoncollins, 2021-03-29:
I will try that search when I get back to my build machine.  In the meantime though, those entries seem to be invisible intentionally, by their lack of prompt text:

```
config IIO_BUFFER_DMA
	tristate
	help
	  Provides the generic IIO DMA buffer infrastructure that can be used by
	  drivers for devices with DMA support to implement the IIO buffer.

	  Should be selected by drivers that want to use the generic DMA buffer
	  infrastructure.

config IIO_BUFFER_DMAENGINE
	tristate
	select IIO_BUFFER_DMA
	help
	  Provides a bonding of the generic IIO DMA buffer infrastructure with the
	  DMAengine framework. This can be used by converter drivers with a DMA port
	  connected to an external DMA controller which is supported by the
	  DMAengine framework.

	  Should be selected by drivers that want to use this functionality.
```

jamesoncollins, 2021-03-29:
Here is the output of a search for `IIO_BUFFER_DMA`:

```
  │ Symbol: IIO_BUFFER_DMA [=m]                                                                                                                                                     │  
  │ Type  : tristate                                                                                                                                                                │  
  │   Defined at drivers/iio/buffer/Kconfig:12                                                                                                                                      │  
  │   Depends on: IIO [=y] && IIO_BUFFER [=y]                                                                                                                                       │  
  │   Selected by [m]:                                                                                                                                                              │  
  │   - IIO_BUFFER_DMAENGINE [=m] && IIO [=y] && IIO_BUFFER [=y]                                                                                                                    │  
  │   Selected by [n]:                                                                                                                                                              │  
  │   - IIO_ALL_ADI_DRIVERS [=n]                                                                                                                                                    │  
  │                                                                                                                                                                                 │  
  │                                                                                                                                                                                 │  
  │ Symbol: IIO_BUFFER_DMAENGINE [=m]                                                                                                                                               │  
  │ Type  : tristate                                                                                                                                                                │  
  │   Defined at drivers/iio/buffer/Kconfig:21                                                                                                                                      │  
  │   Depends on: IIO [=y] && IIO_BUFFER [=y]                                                                                                                                       │  
  │   Selects: IIO_BUFFER_DMA [=m]                                                                                                                                                  │  
  │   Selected by [m]:                                                                                                                                                              │  
  │   - MWIPCORE_IIO_STREAMING [=m] && MATHWORKS_IP_CORE [=m] && MWIPCORE [=m] && DMADEVICES [=y] && IIO [=y]                                                                       │  
  │   Selected by [n]:                                                                                                                                                              │  
  │   - IIO_ALL_ADI_DRIVERS [=n]                                                                                                                                                    │  
  │   - CF_AXI_ADC [=n] && IIO [=y]                                                                                                                                                 │  
  │   - CF_AXI_DDS [=n] && IIO [=y] && SPI [=y] && COMMON_CLK [=y]  
```

jamesoncollins, 2021-03-29:
Setting all of the ADI options on the main menuconfig page to =y will mark these two packages as =y  

```
  │ Symbol: IIO_BUFFER_DMA [=y]                                                                                                                                                     │  
  │ Type  : tristate                                                                                                                                                                │  
  │   Defined at drivers/iio/buffer/Kconfig:12                                                                                                                                      │  
  │   Depends on: IIO [=y] && IIO_BUFFER [=y]                                                                                                                                       │  
  │   Selected by [y]:                                                                                                                                                              │  
  │   - IIO_ALL_ADI_DRIVERS [=y]                                                                                                                                                    │  
  │   - IIO_BUFFER_DMAENGINE [=y] && IIO [=y] && IIO_BUFFER [=y]                                                                                                                    │  
  │                                                                                                                                                                                 │  
  │                                                                                                                                                                                 │  
  │ Symbol: IIO_BUFFER_DMAENGINE [=y]                                                                                                                                               │  
  │ Type  : tristate                                                                                                                                                                │  
  │   Defined at drivers/iio/buffer/Kconfig:21                                                                                                                                      │  
  │   Depends on: IIO [=y] && IIO_BUFFER [=y]                                                                                                                                       │  
  │   Selects: IIO_BUFFER_DMA [=y]                                                                                                                                                  │  
  │   Selected by [y]:                                                                                                                                                              │  
  │   - IIO_ALL_ADI_DRIVERS [=y]                                                                                                                                                    │  
  │   - CF_AXI_ADC [=y] && IIO [=y]                                                                                                                                                 │  
  │   - CF_AXI_DDS [=y] && IIO [=y] && SPI [=y] && COMMON_CLK [=y]                                                                                                                  │  
  │   Selected by [m]:                                                                                                                                                              │  
  │   - MWIPCORE_IIO_STREAMING [=m] && MATHWORKS_IP_CORE [=m] && MWIPCORE [=m] && DMADEVICES [=y] && IIO [=y] 
```

mirceacaprioru, 2021-03-31:
Hi,

Thanks for catching this. Sometimes we miss dependencies like this because of ADI options that select a lot of them by default. 

Do you want to do a pull request or we can fix it?

Regards,
Mircea

jamesoncollins, 2021-03-31:
Sure.  The AD7768-1 has the same issue, should they both be updated?

Something like:

```
config AD7768
	tristate "Analog Devices AD7768 ADC driver"
	depends on SPI
+	select IIO_BUFFER
+	select IIO_BUFFER_DMA
+	select IIO_BUFFER_DMAENGINE
	help
	  Say yes here to build support for Analog Devices AD7768 SPI analog to
	  digital converter (ADC).

	  To compile this driver as a module, choose M here: the module will be
	  called ad7768.

config AD7768_1
	tristate "Analog Devices AD7768-1 ADC driver"
	depends on SPI
	select IIO_BUFFER
	select IIO_TRIGGER
	select IIO_TRIGGERED_BUFFER
+	select IIO_BUFFER_DMA
+	select IIO_BUFFER_DMAENGINE
	help
	  Say yes here to build support for Analog Devices AD7768-1 SPI
	  simultaneously sampling sigma-delta analog to digital converter (ADC).

	  To compile this driver as a module, choose M here: the module will be
	  called ad7768-1.
```

mirceacaprioru, 2021-03-31:
yes that would work. Thanks again!

mirceacaprioru, 2021-04-01:
Fixed with https://github.com/analogdevicesinc/linux/pull/1497

closed 2021-04-01 by mirceacaprioru