analogdevicesinc/linux #1501: incorrect dma address range

jamesoncollins, 2021-04-07:
https://github.com/analogdevicesinc/linux/blob/83ce6e3b1894d512c1c41ae0391bad8f46c10343/arch/arm/boot/dts/zynq-zed-adv7511-ad7768.dts#L38

According to [this post](https://ez.analog.com/members/jay_5f00_col/activities) the address range for this device should be 0x1000.

It doesn't appear to break anything by being incorrect though.

mirceacaprioru, 2021-04-08:
Hi,

You are right it seems to be a typo. The question again is do you want to make a PR or should we do it?

Thanks again for catching this.

Mircea

jamesoncollins, 2021-04-14:
Sorry for the late reply, sure.

But should it be for this one file, or all occurrences of "adi,axi-dmac-1.00.a" with a range of 0x10000?

I believe that this is a complete list:

```
zynq-zc706-adv7511-ad9434-fmc-500ebz.dts-		reg = <0x44a30000 0x10000>;
zynq-zed-imageon.dts-			reg = <0x43000000 0x10000>;
zynq-zed-imageon.dts-			reg = <0x43c20000 0x10000>;
zynq-zed-adv7511-ad7768-axi-adc.dts-		reg = <0x7c480000 0x10000>;
zynq-zc702-adv7511.dtsi-			reg = <0x43000000 0x10000>;
zynq-zed-adv7511-ad9467-fmc-250ebz.dts-		reg = <0x44A30000 0x10000>;
zynq-zc706-adv7511-ad9625-fmcadc2.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-fmcomms1.dts-		reg = <0x7c400000 0x10000>;
zynq-zed-adv7511-fmcomms1.dts-		reg = <0x7c420000 0x10000>;
adi-zynq-cn0363.dtsi-		reg = <0x44a30000 0x10000>;
zynq-zc706-adv7511-ad9265-fmc-125ebz.dts-		reg = <0x44a30000 0x10000>;
zynq-zc706-adv7511-fmcdaq3.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcdaq3.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-fmcomms11-RevA.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcomms11-RevA.dts-		reg = <0x7c420000 0x10000>;
zynq-adrv9002-rx2tx2.dtsi-			reg = <0x44A50000 0x10000>;
zynq-zed-adv7511.dtsi-			reg = <0x43000000 0x10000>;
zynq-zc706-adv7511-fmcdaq2.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcdaq2.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-adrv9008-1.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-adrv9371.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-adrv9371.dts-		reg = <0x7c440000  0x10000>;
zynq-zc706-adv7511-adrv9371.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-ad9739a-fmc.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c400000 0x10000>;
zynq-zed-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-adrv9009.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-adrv9009.dts-		reg = <0x7c440000  0x10000>;
zynq-zc706-adv7511-adrv9009.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-fmcjesdadc1.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-fmcjesdadc1.dts-		reg = <0x7c430000 0x10000>;
zynq-zc706-adv7511-ad9361-fmcomms5.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-ad9361-fmcomms5.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-fmcmotcon2.dts-		reg = <0x40510000 0x10000>;
zynq-zed-adv7511-fmcmotcon2.dts-		reg = <0x40520000 0x10000>;
zynq-zed-adv7511-fmcmotcon2.dts-		reg = <0x40540000 0x10000>;
zynq-zed-adv7511-fmcmotcon2.dts-		reg = <0x40550000 0x10000>;
zynq-zed-adv7511-m2k-revb.dts-		reg = <0x7c440000 0x10000>;
zynq-zed-adv7511-m2k-revb.dts-		reg = <0x7c460000 0x10000>;
zynq-zed-adv7511-m2k-revb.dts-		reg = <0x7c480000 0x10000>;
zynq-zed-adv7511-m2k-revb.dts-		reg = <0x7c400000 0x10000>;
zynq-zed-adv7511-m2k-revb.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-ad9172-fmc-ebz.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-ad7768.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcomms6.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511.dtsi-			reg = <0x43000000 0x10000>;
zynq-zc706-adv7511-fmcadc4.dts-		reg = <0x7c400000 0x10000>;
zynq-adrv9361-z7035-fmc.dts-			reg = <0x43000000 0x10000>;
zynq-zc702-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c400000 0x10000>;
zynq-zc702-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c420000 0x10000>;
zynq-adrv9002.dtsi-			reg = <0x44A40000 0x10000>;
zynq-adrv9002.dtsi-			reg = <0x44A50000 0x10000>;
zynq-adrv9002.dtsi-			reg = <0x44A60000 0x10000>;
zynq-zc706-adv7511-ad9625-fmcadc7.dts-		reg = <0x7c420000 0x10000>;
zynq-zc702-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c400000 0x10000>;
zynq-zc702-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-adrv9008-2.dts-		reg = <0x7c440000 0x10000>;
zynq-zc706-adv7511-adrv9008-2.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-ad9361-fmcomms2-3.dts-		reg = <0x7c420000 0x10000>;
zynq-zc702-adv7511-fmcomms1.dts-		reg = <0x7c400000 0x10000>;
zynq-zc702-adv7511-fmcomms1.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-ad9081.dts-			reg = <0x7c420000 0x10000>;
[... 11 lines cut ...]
zynq-e310.dts-			reg = <0x7c400000 0x10000>;
zynq-e310.dts-			reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-ad7768-1-evb.dts-		reg = <0x44a30000 0x10000>;
zynq-zc706-adv7511-ad6676-fmc.dts-		reg = <0x7c420000 0x10000>;
zynq-adrv9361-z7035.dtsi-			reg = <0x7c400000 0x10000>;
zynq-adrv9361-z7035.dtsi-			reg = <0x7c420000 0x10000>;
zynq-zc702-adv7511-ad9361-fmcomms5.dts-		reg = <0x7c400000 0x10000>;
zynq-zc702-adv7511-ad9361-fmcomms5.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c400000 0x10000>;
zynq-zed-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-ad9364-fmcomms4.dts-		reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-fmcomms1.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcomms1.dts-		reg = <0x7c420000 0x10000>;
zynq-adrv9364-z7020.dtsi-			reg = <0x7c400000 0x10000>;
zynq-adrv9364-z7020.dtsi-			reg = <0x7c420000 0x10000>;
zynq-zc706-adv7511-fmcomms11.dts-		reg = <0x7c400000 0x10000>;
zynq-zc706-adv7511-fmcomms11.dts-		reg = <0x7c420000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c440000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c460000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c480000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c460000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c400000 0x10000>;
zynq-zed-adv7511-m2k-revc.dts-		reg = <0x7c420000 0x10000>;
```

mirceacaprioru, 2021-04-19:
Sorry for the delay, i think you are right this is the list with the wrong typo. I would suggest a PR with all of these fixes. 

Thanks again for the help.

iamkkushal, 2022-10-03:
I am working on this issue. Any further inputs rather than @jamesoncollins comment?

PR #2021 (2022-10-06): Fix incorrect DMA address range

Closes #1501
commit db226857 (2022-10-06): arch: arm: dts: fix incorrect DMA address range

Link to documentation explaining the correct address range:
https://ez.analog.com/members/jay_5f00_col/activities

Closes #1501

(cherry picked from commit c78f59d0f789f5c9d29a6f939057620607ee0c62)
commit 6b7929f6 (2022-10-06): arch: arm: dts: fix incorrect DMA address range

Link to documentation explaining the correct address range:
https://ez.analog.com/members/jay_5f00_col/activities

Closes #1501

(cherry picked from commit c78f59d0f789f5c9d29a6f939057620607ee0c62)
commit c78f59d0 (2022-10-06): arch: arm: dts: fix incorrect DMA address range

Link to documentation explaining the correct address range:
https://ez.analog.com/members/jay_5f00_col/activities

Closes #1501
closed 2022-10-06 by nunojsa