analogdevicesinc/linux #1583: What is the best practice to access ad9361_set_tx_atten, ad9361_get_tx_atten and ad9361_ctrl_outs_setup?

JiaoXianjun, 2021-06-23:
Hello,

This is Xianjun from the openwifi project (https://github.com/open-sdr/openwifi). We have built a WiFi "chip" in FPGA , and support several types of  AD936x SDR boards (Xilinx FPGA) as you can see in our github.

Currently, to do real-time tx att control (response to the radio control of rfkill) and read real-time AGC gain to the baseband/FPGA (for RSSI calculation required by WiFi CCA -- Clear Channel Assessment), we have a modified ad9361.c (only add 3 EXPORT_SYMBOL) to give direct access of ad9361_set_tx_atten/ad9361_get_tx_atten/ad9361_ctrl_outs_setup from our WiFi driver: https://github.com/open-sdr/openwifi/blob/master/driver/sdr.c

Would you please advice us: Is there a better practice than maintaining our own version of ad9361.c (with 3 additional EXPORT_SYMBOL in it)? Any method to avoid adding EXPORT_SYMBOL?

Thanks!

tfcollins, 2021-06-24:
You shouldn't have to modify the driver to control these functions. ctrl_out config is accessible from the driver's debug attributes without modification or just through device tree

https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361-customization?&#control_output_setup

Attenuation control is available through standard attributes: https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361#tx_attenuation_control

JiaoXianjun, 2021-07-06:
Thanks!
I will try and let you know.