analogdevicesinc/linux #1598: clk_notifier usage in drivers

mhennerich, 2021-07-02:
https://github.com/analogdevicesinc/linux/blob/c588ee4bed9acde993a807188044c6befe2c0ef1/drivers/iio/frequency/admv1013.c#L350

and also here: https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/frequency/admv1014.c#L422

Second arg is action and not flag!
https://elixir.bootlin.com/linux/v5.4.129/source/include/linux/notifier.h#L51

There are 3 types of events - so if we don't want to update twice - we should just use POST_RATE_CHANGE or PRE_RATE_CHANGE  and handle ABORT_RATE_CHANGE

```
/**
 * DOC: clk notifier callback types
 *
 * PRE_RATE_CHANGE - called immediately before the clk rate is changed,
 *     to indicate that the rate change will proceed.  Drivers must
 *     immediately terminate any operations that will be affected by the
 *     rate change.  Callbacks may either return NOTIFY_DONE, NOTIFY_OK,
 *     NOTIFY_STOP or NOTIFY_BAD.
 *
 * ABORT_RATE_CHANGE: called if the rate change failed for some reason
 *     after PRE_RATE_CHANGE.  In this case, all registered notifiers on
 *     the clk will be called with ABORT_RATE_CHANGE. Callbacks must
 *     always return NOTIFY_DONE or NOTIFY_OK.
 *
 * POST_RATE_CHANGE - called after the clk rate change has successfully
 *     completed.  Callbacks must always return NOTIFY_DONE or NOTIFY_OK.
 *
 */
#define PRE_RATE_CHANGE			BIT(0)
#define POST_RATE_CHANGE		BIT(1)
#define ABORT_RATE_CHANGE		BIT(2)
```

amiclaus, 2021-07-12:
@mhennerich the patches were merged to master. I believe this issue can be closed.

mhennerich, 2021-07-12:
Thanks! - Yes - Closed

closed 2021-07-12 by mhennerich