analogdevicesinc/linux #1751: README incorrectly says latest Xilinx kernel is 4.14

mhatle, 2021-12-02:
The README says:
             The current version of upstream Linux that Xilinx merged into 
             their tree is from Linux 4.14. No other patches seem to have been 
             added after this (by Xilinx).

This is not true, the current Xilinx kernel, available at:
    https://github.com/xilinx/linux-xlnx

Is based on 5.10.0.  Additional prior releases 4.19 and 5.4 are also available.

aholtzma-am, 2021-12-17:
Any plans to merge in the upstream kernel from Xilinx? 4.14 is extremely old at this point.

mhennerich, 2021-12-17:
I guess readme needs updating but we're always close. If you check were at 5.10 and have merged 2021.1 from Xilinx.

mhennerich, 2022-02-24:
This can be closed now.

closed 2022-02-24 by mhennerich