analogdevicesinc/linux #1986: buildbreak in master: /include/linux/linkage.h:6:10: fatal error: linux/stringify.h: No such file or directory

catkira, 2022-08-26:
When compiling with the master branch I get this error:
```
In file included from ./arch/arm/include/asm/bug.h:5,
                 from ./include/linux/bug.h:5,
                 from ./include/linux/page-flags.h:10,
                 from kernel/bounds.c:10:
./include/linux/linkage.h:6:10: fatal error: linux/stringify.h: No such file or directory
    6 | #include <linux/stringify.h>
      |          ^~~~~~~~~~~~~~~~~~~
```

catkira, 2022-09-01:
I solved this problem by not using the latest master branch. I was compiling with Vitis 2022.1.

I am currently using [this](https://github.com/catkira/linux/tree/antsdr) linux with Vitis 2022.1.

nunojsa, 2022-09-05:
Hmm that's odd... Anyways, I'm closing this one as I cannot reproduce it and Vitis 2022.1 (and the compiler it brings) was never tested and not expected to reliably work.

closed 2022-09-05 by nunojsa

catkira, 2022-09-05:
ok, I will get back to You if I run into this problem again.