analogdevicesinc/linux #2239: Extend support for AD40xx with variable sample frequency

pamolloy, 2023-07-27:


pdp7, 2023-08-14:
YAML conversion of the bindings is in 
https://github.com/machschmitt/linux/tree/dev/adaq4003
Patch with [adi,ad400x.yaml](https://github.com/analogdevicesinc/linux/pull/2171/commits/bb207a18ba846a642d1d18604ed14ec06ab213d0#diff-4a05ce3215f0ab3d788a66b149ac3e84e641d6c8580ea9439c06bf7c46479ece)

pdp7, 2023-08-14:
Internal MR [iio: adc: ad400x: Add additional supported parts](https://gitlab.baylibre.com/baylibre/analog-devices-adi/linux/-/merge_requests/3)
Sampling frequency changes will be in separate internal MR.

pdp7, 2023-08-19:
New BOOT.BIN to enable PWM device for the CNV signal is now available in https://github.com/adi-ses/sea-misc/pull/27 and will be tested

pdp7, 2023-08-19:
I've created a merge request for internal review which adds support in the Linux ad400x driver to get and set the sampling frequency.

pdp7, 2023-08-23:
FYI - the updated BOOT.BIN from https://github.com/adi-ses/sea-misc/pull/27#issuecomment-1689283610 is breaking SPI.

pdp7, 2023-08-31:
FYI - my changes conflict with https://github.com/analogdevicesinc/linux/pull/2171

pdp7, 2023-09-15:
@mhennerich I was told recently by @pamolloy that we should target upstream for the iio driver changes such as my PR #2257.  

The [ad400x.c](https://github.com/analogdevicesinc/linux/commits/master/drivers/iio/adc/ad400x.c) driver is not upstream.  It was introduced in https://github.com/analogdevicesinc/linux/commit/24878df90307fb41596d5df04233ac6a183541ac by @commodo and most recently a small change by @nunojsa in https://github.com/analogdevicesinc/linux/commit/24878df90307fb41596d5df04233ac6a183541ac.  

Should I upstream the ad400x iio driver?

If so, should I start with a patch series containing the current 5 commits in adi master rebased on v6.6-rc1?  

I would also include the [yaml bindings](https://github.com/analogdevicesinc/linux/pull/2171/files#diff-4a05ce3215f0ab3d788a66b149ac3e84e641d6c8580ea9439c06bf7c46479ece) created by @machschmitt in PR #2171

pdp7, 2023-09-18:
**_Details on the compile errors for ad400x.c in mainline._**

### Issue: iio_dmaengine_buffer_submit_block
```
drivers/iio/adc/ad400x.c:323:16: error: implicit declaration of function 'iio_dmaengine_buffer_submit_block' [-Werror=implicit-function-declaration]
  323 |         return iio_dmaengine_buffer_submit_block(queue, block, DMA_DEV_TO_MEM);
```
This is because in mainline:
```
drivers/iio/buffer/industrialio-buffer-dmaengine.c:static int iio_dmaengine_buffer_submit_block(struct iio_dma_buffer_queue *queue,
```
However, ADI master removes 'static' in this merge commit:
```
commit fe7ded91d236c729703342214c214955224ae8e5
Merge: 43000ae1c6f2 4dcc2058b862
Author: Lars-Peter Clausen <lars@metafoo.de>
Date:   Mon Aug 1 18:44:35 2016 +0200

    Merge remote-tracking branch 'xilinx/master' into xcomm_zynq_4_6
where is just removes 'static'
diff --cc drivers/iio/buffer/industrialio-buffer-dmaengine.c
index f395a10c5f9e,9fabed47053d..31480bbca319
--- a/drivers/iio/buffer/industrialio-buffer-dmaengine.c
+++ b/drivers/iio/buffer/industrialio-buffer-dmaengine.c
@@@ -41,82 -54,50 +55,65 @@@ static void iio_dmaengine_buffer_block_
        iio_dma_buffer_block_done(block);
  }
  
- int iio_dmaengine_buffer_submit_block(struct iio_dma_buffer_block *block,
-       int direction)
 -static int iio_dmaengine_buffer_submit_block(struct iio_dma_buffer_queue *queue,
 -      struct iio_dma_buffer_block *block)
++int iio_dmaengine_buffer_submit_block(struct iio_dma_buffer_queue *queue,
++      struct iio_dma_buffer_block *block, int direction)
```

### Issue: 'struct iio_dma_buffer_block' has no member named 'block'
```
drivers/iio/adc/ad400x.c: In function 'hw_submit_block':
drivers/iio/adc/ad400x.c:321:14: error: 'struct iio_dma_buffer_block' has no member named 'block'
  321 |         block->block.bytes_used = block->block.size;
      |              ^~
drivers/iio/adc/ad400x.c:321:40: error: 'struct iio_dma_buffer_block' has no member named 'block'
  321 |         block->block.bytes_used = block->block.size;
      |                                        ^~
```
This is due to:
https://lore.kernel.org/all/20211115141925.60164-3-paul@crapouillou.net/
```
commit ed14e769f64311769dcf20dde544b82c158d01b1
Author: Paul Cercueil <paul@crapouillou.net>
Date:   Mon Nov 15 14:19:12 2021 +0000

    iio: buffer-dma: Remove unused iio_buffer_block struct
    
    This structure was never used anywhere, so it can safely be dropped.
    
    It will later be re-introduced as a different structure in a
    different header.
    
    Signed-off-by: Paul Cercueil <paul@crapouillou.net>
    Reviewed-by: Alexandru Ardelean <ardeleanalex@gmail.com>
    Link: https://lore.kernel.org/r/20211115141925.60164-3-paul@crapouillou.net
    Signed-off-by: Jonathan Cameron <Jonathan.Cameron@huawei.com>

diff --git a/include/linux/iio/buffer-dma.h b/include/linux/iio/buffer-dma.h
index ff15c61bf319..6564bdcdac66 100644
--- a/include/linux/iio/buffer-dma.h
+++ b/include/linux/iio/buffer-dma.h
@@ -17,11 +17,6 @@ struct iio_dma_buffer_queue;
[... 70 lines cut ...]
        The in-kernel interface is currently specific to the SPI-Engine controller
        and only works for certain use cases, in the future it will be extended to
        be more generic and support more usecases.
    
        Signed-off-by: Lars-Peter Clausen <lars@metafoo.de>
```
In mainline, Lars-Peter Clausen later renamed spi-engine to spi-axi-spi-engine but the header only ever existed in the adi downstream.
```
commit 4dbaf1c9ecd93c7767ebe33759b14d36bb21bc78
Author: Lars-Peter Clausen <lars@metafoo.de>
Date:   Thu Feb 4 16:48:51 2016 +0100

    spi: spi-engine: Sync with upstream
    
    Sync the AXI SPI Engine with the version of the driver that went upstream.
    No functional changes, just moving things slightly around in a few places.
    
    Signed-off-by: Lars-Peter Clausen <lars@metafoo.de>

 drivers/spi/Kconfig                                |  11 +++--
 drivers/spi/Makefile                               |   2 +-
 drivers/spi/{spi-engine.c => spi-axi-spi-engine.c} | 134 +++++++++++++++++++++++++++++++++--------------------
 3 files changed, 92 insertions(+), 55 deletions(-)
```
I was able to solve this issue by copying over the header file into mainline.  However, it remains to be seen if this is a proper solution or not. (edited)

machschmitt, 2023-09-18:
Hi, adding some thoughts about this task ...

The ad400x driver has a dependency on the [SPI offload module](https://wiki.analog.com/resources/tools-software/linux-drivers/spi/spi_engine) which is not supported upstream yet.
The [offload module](https://wiki.analog.com/resources/fpga/peripherals/spi_engine/offload) is used to store SPI messages with the intent of achieving shorter delay between transactions.
Comparing [mainline](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/spi/spi-axi-spi-engine.c?h=v6.6-rc2) with [adi-master](https://github.com/analogdevicesinc/linux/blob/master/drivers/spi/spi-axi-spi-engine.c) we can see a few differences but most of them are related to the offload module.
I don't know how he could be so sure, but at some point in time Lars has said the "Linux kernel will get SPI offloading support to its standard SPI API" [[1]](https://wiki.analog.com/_media/resources/fpga/peripherals/spi-engine3.pdf).
So, if I had to come up with a plan, I would try to upstream the ad400x along with the SPI engine offload so the offload module was a user to justify its introduction.
Sorry for the delayed reply on this.

@pdp7, since the goal is to upstream the drivers, I think we should merge #2257 and  #2171 so we can send it all in one patch set.
How can we coordinate efforts toward that goal?

pdp7, 2023-09-19:
@machschmitt After I posted the above, @mhennerich told me on teams to finish the PR on the 5.15 kernel before looking at upstreaming.

He suggested that it would be good to coordinate upstreaming with @nunojsa as there is a [buffer-dma series from Paul Cercueil in progress](https://lore.kernel.org/linux-iio/20230807112113.47157-1-paul@crapouillou.net/#r), and he explained that some of the mainline compile errors I noted above are due to the fact that adi/master has buffer write support and mainline does not yet.

mhennerich, 2023-09-26:
> @machschmitt After I posted the above, @mhennerich told me on teams to finish the PR on the 5.15 kernel before looking at upstreaming.
>
> He suggested that it would be good to coordinate upstreaming [...]

@pcercuei  When do you think you sent v5 ?

pcercuei, 2023-09-26:
It's waiting on @nunojsa's work on upstreaming the AXI DAC.

mhennerich, 2023-09-26:
> It's waiting on @nunojsa's work on upstreaming the AXI DAC.

I see there is a dependency chain - @nunojsa when do you think  you can start on this?

nunojsa, 2023-09-27:
> It's waiting on @nunojsa's work on upstreaming the AXI DAC.

Unfortunately that might take some time as I was pulled to another thing so I might only be back on this next week or so... Also, there's some stm folks that want to add some similar infrastructure as we want, so we need to sync and make sure it works for both of us (and that also takes more time than usual).

Anyways, @pcercuei, correct me if I'm wrong but I'm not seeing any blocker in here... This driver is an adc one so upstream should already have support for IIO input DMA buffers. You won't have any high speed interface like our local mmap stuff or Paul's DMABUF (upsreaming in progress) so you need to use the fileio interface but it should be enough to validate the driver (naturally the throughput won't be as good). As for the compile time errors, you can easily bypass them and use the upstream API. Note that I've done it in the [staging master-next branch](https://github.com/analogdevicesinc/linux/blob/staging/master-next/drivers/iio/adc/ad400x.c#L375). You still have an extra parameter for the buffer direction but you can ignore it for this driver as default is input. So, to sum it up, what we are missing usptream (in terms of IIO) is:

1) High speed interface for IIO DMA buffers;
2) Output buffer support for IIO DMA buffers (not that for typical kfifo buffer, output support is already upstream) and this one is the one pending on my work.

But on 2), I would also like to mention that if you guys have any DAC pending on that support, you can just take Paul's series and include it in your patchset. Jonathan made it clear that he just wants an user for it to merge it, if it is the axi-dac or something else it does not really matter :). Fairly sure that @pcercuei does not care either...

And of course, anything that's not clear, you can always reach out to me so we can see what can be done...

pcercuei, 2023-09-27:
> He suggested that it would be good to coordinate upstreaming with @nunojsa as there is a [buffer-dma series from Paul Cercueil in [...]

Is there actually a dependency on buffer write support?

nunojsa, 2023-09-27:
> Is there actually a dependency on buffer write support?

For this driver I don't believe there is since it's an ADC so it should not matter. We can use what there's upstream... Sure, there's no high speed interface but should still be good enough to test this device.

pdp7, 2023-10-08:
FYI - I have rewritten the support for the additional AD40xx parts on top of the ADAQ4003 support that @machschmitt created https://github.com/machschmitt/linux/pull/1. My goal for this PR is to allow @machschmitt and myself to sync up on the support for the AD40xx parts and the ADAQ4003.

pdp7, 2023-10-10:
@nunojsa 
> For this driver I don't believe there is since it's an ADC so it should not matter. We can use what there's upstream... Sure, there's no high speed interface but should still be good enough to test this device.

I've opened this PR that integrates variable sampling frequency support for AD40XX on top @machschmitt's ADAQ4003 branch. I'm waiting for feedback on that, so I revisited the topic of getting ad400x to work on mainline.

I'll try [what you wrote in](https://github.com/analogdevicesinc/linux/issues/2239#issuecomment-1736848406):
> This driver is an adc one so upstream should already have support for IIO input DMA buffers.
> You won't have any high speed interface [..] so you need to use the fileio interface but it should be enough to validate the driver (naturally the throughput won't be as good).
> As for the compile time errors, you can easily bypass them and use the upstream API.

pdp7, 2023-10-12:
> As for the compile time errors, you can easily bypass them and use the upstream API. Note that I've done it in the [staging master-next branch](https://github.com/analogdevicesinc/linux/blob/staging/master-next/drivers/iio/adc/ad400x.c#L375).

@nunojsa 
That link does not work and staging/master-next does not seem to exist anymore.

I looked at these branches with next in their name but they all seem similar to the adi master:
* https://github.com/analogdevicesinc/linux/commits/master-next/drivers/iio/adc/ad400x.c
* https://github.com/analogdevicesinc/linux/blob/master-next/drivers/iio/adc/ad400x.c
* https://github.com/analogdevicesinc/linux/commits/staging/master-next-ci/drivers/iio/adc/ad400x.c

I looked at the 6.1 branch.  I see that ad400x.c is a single commit by @commodo, but I don't see any changes from adi master:
* https://github.com/analogdevicesinc/linux/blob/adi-6.1.0/drivers/iio/adc/ad400x.c

nunojsa, 2023-10-12:
> That link does not work and staging/master-next does not seem to exist anymore.

Yeah, we already merged master-next with master... You can now look at master.

> I looked at the 6.1 branch. I see that ad400x.c is a single commit by @commodo, but I don't see any changes from adi master:

That is a rebased branch and in drivers not upstreamed we typically squash all the patches we have for it in one since when upstreaming you'll want to send just the patch adding support for the device (of course that if the device is really complicated that won't apply).

Also note that you still can't use the API as mainline kernel as we have an extra parameter for the buffer direction. But being this an input device, you can ignore that parameter

pdp7, 2023-10-12:
@nunojsa 
> Also note that you still can't use the API as mainline kernel as we have an extra parameter for the buffer direction. But being this an input device, you can ignore that parameter
Thank you, I've changed the call to devm_iio_dmaengine_buffer_setup() to remove the last argument (IIO_BUFFER_DIRECTION_IN).

I've been trying to understand how to work around `spi_engine_offload_enable()` and `spi_engine_offload_load_msg()` not existing upstream.  I noticed that drivers/iio/adc/adi-axi-adc.c is using DMA but it is not using the Offload.  Would that we a good trying to look at?

nunojsa, 2023-10-13:
> I've been trying to understand how to work around spi_engine_offload_enable() and spi_engine_offload_load_msg() not existing upstream.

That's not possible... I think someone already mentioned in here or some other thread that for this driver, upstreaming the offload engine is also needed. And that is some significant amount of work to do (but maybe a fun one at least). AFAIK, we don't support any design without the offload core.

> I noticed that drivers/iio/adc/adi-axi-adc.c is using DMA but it is not using the Offload. Would that we a good trying to look at?

That one is a completely different story. It's again a driver for a soft core but this one typically connects to high speed devices with CMOS, LVDS or JESD interfaces. On top of that core then we have the DMA core (there's also an upstream driver for it) for passing samples to memory/device (depending on direction). And the IIO DMA interface is just about connecting IIO buffering with DMA.

Note that for spi_engine, you have a spi device (supporting things like spi multiple SDO lines) that interfaces with the FPGA through SPI (where the spi_engine core comes into place). Then, again, on top of it comes the DMA core so we can keep up with high speed sampling.

commodo, 2023-10-13:
> I noticed that drivers/iio/adc/adi-axi-adc.c is using DMA but it is not using the Offload. Would that we a good trying to look at?

The mainline AXI-ADC driver was upstreamed mostly to finalize all the work that Lars did for the IIO DMA buffer :)
Well, that was mostly my intention with it.
A lot of pieces were there already (put by Lars).
The IIO DMA buffer was upstreamed before the AXI ADC driver.
But there was no user of the IIO DMA buffer in mainline (until the AXI ADC driver).
(Which is interesting these days, because in upstream kernel, they don't usually let you put in frameworks before you have a user)

Buf the irony of the AXI ADC, is that it also requires a user.
(There are standalone AXI ADC use-cases, but there weren't any convincing enough FPGA designs to call for upstreaming)
So, the AXI ADC driver was upstreamed as a requirement of the `./drivers/iio/adc/ad9467.c`  driver.
This is also to prepare the way for other high-speed ADCs (and later DACs, when that happens :) )

pdp7, 2023-12-18:
I believe the goal of this PR has been accomplished through the https://github.com/analogdevicesinc/linux/tree/staging/ad400x-add-parts branch that @machschmitt created.  Therefore, I will close this PR.

closed 2023-12-18 by pdp7