analogdevicesinc/linux #2287: kernel oops in spi-engine after read timeout

dlech, 2023-10-16:
I'm working on adding some new parts to the `ad_pulsar.c` driver. There is some issue causing communication to not work so we are hitting the timeout error path in the spi-engine.

https://github.com/analogdevicesinc/linux/blob/a834f68ed763ac56741f06ab69843b297eb718e1/drivers/spi/spi-axi-spi-engine.c#L592-L603

This issue can be reproduced by reading the raw analog value twice. The first time, we get the timeout error from the link above. The second time, we get the kernel crash below. So it appears that something is not getting cleaned up properly after the timout error.

```console
Unable to handle kernel paging request at virtual address ffffffac              
[ffffffac] *pgd=1ffbe861, *pte=00000000, *ppte=00000000                         
Internal error: Oops: 37 [#1] PREEMPT SMP ARM                                   
Modules linked in: 8021q ad_pulsar                                              
CPU: 0 PID: 1227 Comm: iio_attr Not tainted 6.1.0-22219-g190b7577f6c9 #5        
Hardware name: Xilinx Zynq Platform                                             
PC is at spi_engine_rx_next+0x1c/0x74                                           
LR is at spi_engine_transfer_one_message+0xfc/0x170                             
pc : [<c065fa6c>]    lr : [<c066033c>]    psr: a0000093                         
sp : e0be1ce0  ip : 00000007  fp : 00400cc0                                     
r10: 7ffff000  r9 : c1fe088c  r8 : c1fe08c8                                     
r7 : 60000013  r6 : e0be1d80  r5 : 00000000  r4 : c1fe0880                      
r3 : ffffffa8  r2 : 00000000  r1 : e0be1d80  r0 : c1fe0880                      
Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none              
Control: 18c5387d  Table: 02dc004a  DAC: 00000051                               
Register r0 information: slab kmalloc-192 start c1fe0840 pointer offset 64 size2
Register r1 information: 2-page vmalloc region starting at 0xe0be0000 allocated8
Register r2 information: NULL pointer                                           
Register r3 information: non-paged memory                                       
Register r4 information: slab kmalloc-192 start c1fe0840 pointer offset 64 size2
Register r5 information: NULL pointer                                           
Register r6 information: 2-page vmalloc region starting at 0xe0be0000 allocated8
Register r7 information: non-paged memory                                       
Register r8 information: slab kmalloc-192 start c1fe0840 pointer offset 136 siz2
Register r9 information: slab kmalloc-192 start c1fe0840 pointer offset 76 size2
Register r10 information: non-paged memory                                      
Register r11 information: non-paged memory                                      
Register r12 information: non-paged memory                                      
Process iio_attr (pid: 1227, stack limit = 0x72d213d0)                          
Stack: (0xe0be1ce0 to 0xe0be2000)                                               
1ce0: 00000006 f84c0bb9 c2486800 00000001 c2486800 00000000 e0be1d80 c2486800   
1d00: d27ff158 c065b2e4 00400cc0 c0655da4 e0be1dd0 c1c03080 c27ed00c 00000000   
1d20: e0be1d80 c2486a00 c2486800 d3e57f54 d27ff158 7ffff000 00400cc0 c065bc54   
1d40: c14baa80 00000000 00000000 e0be1d4c e0be1d4c f84c0bb9 c2486400 e0be1d80   
1d60: e0be1e6c c4a28008 c4aa5200 c065bdcc e0be1dd0 c2486400 e0be1e6c bf000164   
1d80: e0be1e28 e0be1e28 c2486400 00000000 00000000 00000000 00000004 00000000   
1da0: ffffff8d 00000000 00000000 00000000 e0be1db0 e0be1db0 00000000 f84c0bb9   
1dc0: c1803f10 c4a28340 00003049 bf00054c 00000000 c4a28380 00000001 00000000   
1de0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00100040   
1e00: 00000000 00000000 00000000 04c4b3ff 00000000 00000000 00000000 00000000   
1e20: 00000000 00000000 e0be1d80 e0be1d80 00000000 f84c0bb9 c4a28340 e0be1e6c   
1e40: c64b8f40 bf000674 c01002c4 ca65a580 bf000580 d3e0f000 e0be1e6c c08c313c   
1e60: 00000000 c0297224 00000002 00400cc0 ffffffff c025e108 c02cf194 f84c0bb9   
1e80: d27ff140 c48dd540 d3e0f000 c05d6894 d27ff140 c0e5b9e0 d3e0f000 c0329f90   
1ea0: d27ff140 00004000 e0be1f28 00000001 00000001 c02cf038 652dac9d 00000000   
1ec0: e0be1f10 d27ff168 000033df c02a0a0c 0000002f c2e4f000 ffffffff d3d98600   
1ee0: e0be1f80 c0328618 00000001 00004000 00000000 00004004 00000000 c02a0a0c   
1f00: 00000000 00000000 c2e4f0bc dfc139f4 01000006 00000000 00004000 000398f0   
1f20: 00000000 00000000 d3d98600 00000000 00000000 00000000 00000000 00000000   
1f40: 00000000 00004004 00000000 00000000 652dac9d f84c0bb9 00000003 d3d98600   
1f60: d3d98600 00000000 00000000 c01002c4 ca65a580 00000003 000159c4 c02a140c   
1f80: 00000000 00000000 00000000 f84c0bb9 b6e92c18 00000003 b6f260c0 00004000   
1fa0: 00000003 c0100060 00000003 b6f260c0 00000003 000398f0 00004000 00000000   
1fc0: 00000003 b6f260c0 00004000 00000003 00004000 b6e92780 000398f0 000159c4   
1fe0: fbad2488 be8d9538 b6e0dc94 b6e0dcb0 60000010 00000003 00000000 00000000   
 spi_engine_rx_next from spi_engine_transfer_one_message+0xfc/0x170             
 spi_engine_transfer_one_message from __spi_pump_transfer_message+0x31c/0x62c   
 __spi_pump_transfer_message from __spi_sync+0x318/0x464                        
 __spi_sync from spi_sync+0x2c/0x44                                             
 spi_sync from spi_sync_transfer.constprop.0+0x68/0x90 [ad_pulsar]              
 spi_sync_transfer.constprop.0 [ad_pulsar] from ad_pulsar_read_channel+0x8c/0xc]
 ad_pulsar_read_channel [ad_pulsar] from ad_pulsar_read_raw+0xf4/0x1c0 [ad_puls]
 ad_pulsar_read_raw [ad_pulsar] from iio_read_channel_info+0xec/0x100           
 iio_read_channel_info from dev_attr_show+0x20/0x50                             
 dev_attr_show from sysfs_kf_seq_show+0x94/0x124                                
 sysfs_kf_seq_show from seq_read_iter+0x1b0/0x508                               
 seq_read_iter from vfs_read+0x1d0/0x298                                        
 vfs_read from ksys_read+0x68/0xec                                              
 ksys_read from ret_fast_syscall+0x0/0x54                                       
Exception stack(0xe0be1fa8 to 0xe0be1ff0)                                       
1fa0:                   00000003 b6f260c0 00000003 000398f0 00004000 00000000   
1fc0: 00000003 b6f260c0 00004000 00000003 00004000 b6e92780 000398f0 000159c4   
1fe0: fbad2488 be8d9538 b6e0dc94 b6e0dcb0                                       
Code: e5901014 ea000004 e3530000 0a00000a (e5932004)                            
---[ end trace 0000000000000000 ]---                                            
note: iio_attr[1227] exited with preempt_count 1    
```

mhennerich, 2023-10-17:
Hi David,
Can you try to dig a bit deeper in spi_engine_rx_next() and see what pointer is causing the issue?

dlech, 2023-10-17:
I enabled ftrace and added `noinline` to all of the functions in the SPI Engine driver to trace the code path to compare working vs. the crash. I used `iio_attr ad7985 -ci voltage0-voltage1` to trigger an SPI transaction.

First, here is what a normal transaction looks like:

```console
root@zed-3:~# cat /sys/kernel/tracing/trace
# tracer: function
#
# entries-in-buffer/entries-written: 46/46   #P:2
#
#                                _-----=> irqs-off/BH-disabled
#                               / _----=> need-resched
#                              | / _---=> hardirq/softirq
#                              || / _--=> preempt-depth
#                              ||| / _-=> migrate-disable
#                              |||| /     delay
#           TASK-PID     CPU#  |||||  TIMESTAMP  FUNCTION
#              | |         |   |||||     |         |
        iio_attr-1170    [000] ...1.   129.119160: spi_engine_transfer_one_message <-__spi_pump_transfer_message
        iio_attr-1170    [000] ...1.   129.119167: spi_engine_compile_message <-spi_engine_transfer_one_message
        iio_attr-1170    [000] ...1.   129.119168: spi_engine_get_config <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119170: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119172: spi_engine_get_clk_div <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119175: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119177: spi_engine_get_word_length.constprop.0 <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119178: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119180: spi_engine_gen_cs <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119181: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119182: spi_engine_update_xfer_len <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119183: spi_engine_gen_xfer <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119185: spi_engine_program_add_cmd <-spi_engine_gen_xfer
        iio_attr-1170    [000] ...1.   129.119186: spi_engine_gen_sleep.constprop.0 <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119188: spi_engine_gen_cs <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119189: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119191: spi_engine_compile_message <-spi_engine_transfer_one_message
        iio_attr-1170    [000] ...1.   129.119193: spi_engine_get_config <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119194: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119195: spi_engine_get_clk_div <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119197: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119198: spi_engine_get_word_length.constprop.0 <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119199: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119201: spi_engine_gen_cs <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119202: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119203: spi_engine_update_xfer_len <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119204: spi_engine_gen_xfer <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119206: spi_engine_program_add_cmd <-spi_engine_gen_xfer
        iio_attr-1170    [000] ...1.   129.119207: spi_engine_gen_sleep.constprop.0 <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119209: spi_engine_gen_cs <-spi_engine_compile_message
        iio_attr-1170    [000] ...1.   129.119210: spi_engine_program_add_cmd <-spi_engine_compile_message
        iio_attr-1170    [000] d..2.   129.119211: spi_engine_program_add_cmd <-spi_engine_transfer_one_message
        iio_attr-1170    [000] d..2.   129.119213: spi_engine_write_cmd_fifo <-spi_engine_transfer_one_message
        iio_attr-1170    [000] d..2.   129.119215: spi_engine_tx_next <-spi_engine_transfer_one_message
        iio_attr-1170    [000] d..2.   129.119216: spi_engine_xfer_next <-spi_engine_tx_next
        iio_attr-1170    [000] d..2.   129.119218: spi_engine_xfer_next <-spi_engine_tx_next
        iio_attr-1170    [000] d..2.   129.119219: spi_engine_write_tx_fifo <-spi_engine_transfer_one_message
        iio_attr-1170    [000] d..2.   129.119221: spi_engine_rx_next <-spi_engine_transfer_one_message
        iio_attr-1170    [000] d..2.   129.119222: spi_engine_xfer_next <-spi_engine_rx_next
        iio_attr-1170    [000] d.h1.   129.119229: spi_engine_irq <-__handle_irq_event_percpu
        iio_attr-1170    [000] d.h2.   129.119232: spi_engine_read_rx_fifo <-spi_engine_irq
        iio_attr-1170    [000] d.h2.   129.119233: spi_engine_read_buff <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119235: spi_engine_rx_next <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119236: spi_engine_xfer_next <-spi_engine_rx_next
[... cut ...]
        iio_attr-1166    [000] d..2.    99.727322: spi_engine_tx_next <-spi_engine_transfer_one_message
        iio_attr-1166    [000] d..2.    99.727323: spi_engine_xfer_next <-spi_engine_tx_next
        iio_attr-1166    [000] d..2.    99.727324: spi_engine_xfer_next <-spi_engine_tx_next
        iio_attr-1166    [000] d..2.    99.727326: spi_engine_write_tx_fifo <-spi_engine_transfer_one_message
        iio_attr-1166    [000] d..2.    99.727328: spi_engine_rx_next <-spi_engine_transfer_one_message
        iio_attr-1166    [000] d..2.    99.727329: spi_engine_xfer_next <-spi_engine_rx_next
```

dlech, 2023-10-17:
From the traces above, we can see that in the working case, everything is the same up to:

```console
        iio_attr-1170    [000] d.h1.   129.119229: spi_engine_irq <-__handle_irq_event_percpu
        iio_attr-1170    [000] d.h2.   129.119232: spi_engine_read_rx_fifo <-spi_engine_irq
        iio_attr-1170    [000] d.h2.   129.119233: spi_engine_read_buff <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119235: spi_engine_rx_next <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119236: spi_engine_xfer_next <-spi_engine_rx_next
        iio_attr-1170    [000] d.h2.   129.119238: spi_engine_complete_message <-spi_engine_irq
          <idle>-0       [000] ..s2.   134.152771: spi_engine_timeout <-call_timer_fn
```

In this call tree, `spi_engine_rx_next()` sets `spi_engine->rx_xfer`  to the next xfer if one is available or NULL if there are no more queued xfers. Then `spi_engine_complete_message()` frees the previous xfer by calling `spi_finalize_current_message()`.

In the case of the timeout, `spi_engine_complete_message()` is called but `spi_engine->rx_xfer` is still pointing to memory that gets freed by `spi_engine_complete_message()` via the call to `spi_engine_complete_message()` .

```console
      PK-Backend-1152    [000] ..s1.    66.472911: spi_engine_timeout <-call_timer_fn
      PK-Backend-1152    [000] ..s2.    66.472946: spi_engine_complete_message <-spi_engine_timeout
````

Then the next time we try to do an SPI xfer, the call to `spi_engine_rx_next()` (via `spi_engine_transfer_one_message()`) calls `spi_engine_xfer_next()`  with `spi_engine->rx_xfer` that is still pointing to memory that was freed during the last `spi_engine_complete_message()`.

```console
        iio_attr-1166    [000] d..2.    99.727328: spi_engine_rx_next <-spi_engine_transfer_one_message
        iio_attr-1166    [000] d..2.    99.727329: spi_engine_xfer_next <-spi_engine_rx_next
```

`spi_engine_xfer_next()` accesses the member `xfer->transfer_list` of the freed memory and causes the crash.

dlech, 2023-10-17:
Unrelated, but I noticed another odd thing when everything is working properly (no timeout): the watchdog timer doesn't seem to be canceled on successful completion and still fires 5 seconds afterwards.

```console
        iio_attr-1170    [000] d.h2.   129.119232: spi_engine_read_rx_fifo <-spi_engine_irq
        iio_attr-1170    [000] d.h2.   129.119233: spi_engine_read_buff <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119235: spi_engine_rx_next <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119236: spi_engine_xfer_next <-spi_engine_rx_next
        iio_attr-1170    [000] d.h2.   129.119238: spi_engine_complete_message <-spi_engine_irq
          <idle>-0       [000] ..s2.   134.152771: spi_engine_timeout <-call_timer_fn
```

mhennerich, 2023-10-18:
This sounds odd. I would not expect the timer to be terminated after del_timer().

CONFIG_DEBUG_OBJECTS_TIMERS=y ?

nunojsa, 2023-10-18:
> del_timer() is called immediately before spi_engine_complete_message() so I would expect it to be canceled and not call the callback.

I guess it might depend on concurrency... del_timer() does not care about the callback being scheduled in another CPU... It just deactivates the timer.

```
 Return:
 * * %0 - The timer was not pending
 * * %1 - The timer was pending and deactivated
 ```

The above are the return codes... I wonder if we get 1 in that trace. I'm thinking del_timer_sync() might give us the expected trace?

nunojsa, 2023-10-18:
> I'm thinking del_timer_sync() might give us the expected trace?

hmm, actually looking at the code, this could give us a nasty deadlock...

PR #2292 (2023-10-20): spi: axi-spi-engine: fix use after free after timeout
This fixes a use after free that can happen if the watchdog timer times out on an SPI message then another message is attempted.

The following struct spi_engine members point to memory managed by the spi framework

	struct spi_message *msg;
	struct spi_transfer *tx_xfer;
	const uint8_t *tx_buf;
	struct spi_transfer *rx_xfer;
	uint8_t *rx_buf;

During normal operation, tx_xfer and rx_xfer set to NULL by spi_engine_xfer_next() when the last xfer of a message is completed. However, this code path is not taken when the watchdog timer times out [... cut]
closed 2023-10-20 by nunojsa