analogdevicesinc/linux #2298: axi-spi-engine driver watchdog timer still calls back after being canceled

dlech, 2023-10-20:
Unrelated, but I noticed another odd thing when everything is working properly (no timeout): the watchdog timer doesn't seem to be canceled on successful completion and still fires 5 seconds afterwards.

```console
        iio_attr-1170    [000] d.h2.   129.119232: spi_engine_read_rx_fifo <-spi_engine_irq
        iio_attr-1170    [000] d.h2.   129.119233: spi_engine_read_buff <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119235: spi_engine_rx_next <-spi_engine_read_rx_fifo
        iio_attr-1170    [000] d.h2.   129.119236: spi_engine_xfer_next <-spi_engine_rx_next
        iio_attr-1170    [000] d.h2.   129.119238: spi_engine_complete_message <-spi_engine_irq
          <idle>-0       [000] ..s2.   134.152771: spi_engine_timeout <-call_timer_fn
```

`del_timer()` is called immediately before `spi_engine_complete_message()` so I would expect it to be canceled and not call the callback.

_Originally posted by @dlech in https://github.com/analogdevicesinc/linux/issues/2287#issuecomment-1767315281_

dlech, 2024-01-24:
I think it is fine in the spinlock since the timer should have the `TIMER_IRQSAFE` flag. But if you want to be consistent with the upstream driver, I just put it before the spinlock since it isn't possible for two calls to `spi_engine_transfer_one_message()` to happen at the same time.

nunojsa, 2024-01-24:
Alright,

I'll do this for the 2022_R2 branch and assume for main we well get it in sync with upstream + offload stuff