analogdevicesinc/linux #2323: Potential bug in AD9361 driver

vankxr, 2023-11-03:
https://github.com/analogdevicesinc/linux/blob/2fa4cf1cb0b1036f8c0fca7cc79ccbf27519eea4/drivers/iio/adc/ad9361.c#L5930

There is a call to `ad9361_from_clk` which converts 32-bit stored clock values to 64-bit real frequency values for the RFPLL. However, the value passed into `ad9361_from_clk` is already a 64-bit real frequency value, meaning the function will return double the frequency.

If I am seeing it correctly, the call should be 
`ad9361_calc_rfpll_divder(phy, _rate, parent_rate, &integer, &fract, &vco_div, &vco);`
and not
`ad9361_calc_rfpll_divder(phy, ad9361_from_clk(_rate), parent_rate, &integer, &fract, &vco_div, &vco);`

mhennerich, 2023-11-06:
> There is a call to ad9361_from_clk which converts 32-bit stored clock values to 64-bit real frequency values for the RFPLL. However, the value passed into ad9361_from_clk is already a 64-bit real frequency value, meaning the function will return double the frequency.

Not sure I can follow here -
Depending on the source _rate = phy->current_tx_lo_freq or _rate = phy->current_tx_lo_freq;
current_tx_lo_freq and current_rx_lo_freq are both set by rate, which is the scaled value used in the clock framework.
We do this scaling since 6G is overflowing unsigned long on 32-bit machines.

So ad9361_calc_rfpll_divder() uses correctly ad9361_from_clk(_rate).
What am I missing?

mhennerich, 2024-01-08:
Closed due to inactivity

closed 2024-01-08 by mhennerich