analogdevicesinc/linux #2349: Raspberry build on master

DanielMatyas, 2023-11-16:
I am trying to do a build on master with

`make adi_bcm2711_defconfig; make -j11`

but I keep getting the following error:

> ERROR: modpost: module uio uses symbol dma_buf_attach from namespace DMA_BUF, but does not import it.
ERROR: modpost: module uio uses symbol dma_buf_map_attachment from namespace DMA_BUF, but does not import it.
ERROR: modpost: module uio uses symbol dma_buf_unmap_attachment from namespace DMA_BUF, but does not import it.
ERROR: modpost: module uio uses symbol dma_buf_get from namespace DMA_BUF, but does not import it.
ERROR: modpost: module uio uses symbol dma_buf_put from namespace DMA_BUF, but does not import it.
ERROR: modpost: module uio uses symbol dma_buf_detach from namespace DMA_BUF, but does not import it.
make[1]: *** [scripts/Makefile.modpost:126: Module.symvers] Error 1
make: *** [Makefile:1944: modpost] Error 2

This is most probably due to the merge with the xilinx branch.

I can avoid the error by doing

`make -j11 zImage` - this is not really convinient, because I cannot do a modules_install to the Raspberry in order to use the 6.1 kernel.

or **including** instead of **modularizing** UIO. - this is what I do

Another issue which might be related to this one comes up, when I try to do a build after restarting my computer. I get the following error:

> scripts/gcc-plugins/.arm_ssp_per_task_plugin.so.cmd:1: *** missing separator.  Stop.
make[1]: *** [scripts/Makefile.build:500: scripts/gcc-plugins] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1254: scripts] Error 2
make: *** Waiting for unfinished jobs....

Could you please do something to resolve these issues?

nunojsa, 2023-11-16:
master is not intended to build for raspberry . That's why we have [rpi-6.1.y](https://github.com/analogdevicesinc/linux/tree/rpi-6.1.y) . If that does not build, let me know. Otherwise, close this please

DanielMatyas, 2023-11-20:
Sorry for this. The build works on rpi-6.1.y

closed 2023-11-20 by DanielMatyas