analogdevicesinc/linux #2356: Raspberry build

DanielMatyas, 2023-11-24:
I recently posted an issue with raspberry pi 4 build not working on master, but I closed it, because it was brought to my attention, that these builds should not be done on master: [https://github.com/analogdevicesinc/linux/issues/2349](https://github.com/analogdevicesinc/linux/issues/2349)

On the rpi-6.1.y branch I thought I have no issue, but after a restart of the computer I encountered the same error:

> scripts/gcc-plugins/.arm_ssp_per_task_plugin.so.cmd:1: *** missing separator.  Stop.
make[1]: *** [scripts/Makefile.build:500: scripts/gcc-plugins] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1269: scripts] Error 2
make: *** Waiting for unfinished jobs....

I can fix this by deleting the contents of scripts/gcc-plugins/.arm_ssp_per_task_plugin.so.cmd.

closed 2023-11-24 by nunojsa

Danielvagyok, 2023-11-24:
I did the export, but the changes would take effect only after I delete the contents of scripts/gcc-plugins/.arm_ssp_per_task_plugin.so.cmd

DanielMatyas, 2023-11-24:
I set the cross_compile, but the change only took effect after manually clearing the file scripts/gcc-plugins/.arm_ssp_per_task_plugin.so.cmd