analogdevicesinc/linux #2390: AD9361 “Enable digital saturation gain decrease” setting missing

HavingaThijs, 2024-01-04:
According to the [documentation](https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/ad9361-customization) of the device tree attribute `adi,agc-inner-thresh-high`, this controls whether digital saturation causes a gain decrease when using fast AGC and full table mode (register 0x101 D7). However, it seems what actually gets set is the AGC lock level (0x101 [D6:D0]). 

The macro definition `ENABLE_DIG_SAT_OVRG` seems not to be used anywhere; the driver seems to use register 0x101 (`REG_AGC_LOCK_LEVEL`) only to write the AGC lock level.

I would like to be able to enable this setting using a device tree attribute.

mhennerich, 2024-01-08:
Thanks for reporting - I fixed the wiki page.
And created https://github.com/analogdevicesinc/linux/pull/2391

Regards,
Michael

mhennerich, 2024-01-15:
closed as completed