analogdevicesinc/linux #2397: Make module error

ducanh1699, 2024-01-12:
Hello everyone, I am new to build kernel Image. When I build adrv9009 with option M modulizers in menuconfig I have the error:
Error: modpost:"jesd204 _get_active_links_num" [drivers/iio/adc/ad9009_drv.ko] undefined!
I follow all step in: https://wiki.analog.com/resources/tools-software/linux-drivers/iio-transceiver/adrv9009 
                       and https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842481/Build+kernel
Something worng or missing in my config when making ?
Tks for your answers!

mhennerich, 2024-01-12:
I'll create a fix and PR for this.
In the meantime add:

EXPORT_SYMBOL_GPL(jesd204_get_active_links_num);

underneath the function definition in jesd204-core.c 
Thanks for reporting

mhennerich, 2024-01-15:
Please see: https://github.com/analogdevicesinc/linux/pull/2399

closed 2024-01-15 by mhennerich

ducanh1699, 2024-01-16:
Tks

ducanh1699, 2024-01-19:
I'm sorry but I have another question. Because I just change the adrv9009 to adrv9025 and in the ADRV9025 does not have attribute "profile_config" like adrv9009 so how to change out_voltage_sampling_frequency ? Can you help me about this? Tks ^^