analogdevicesinc/linux #2406: ADRV9025 do not have attribute "profile_config"

ducanh1699, 2024-01-22:
Hello everyone. Because I have just change from adrv9009 to adrv9025, but in adrv9025 I do not see the attribute "profile_config" when I need to change some parameter like "out_voltage_sampling_frequency" and "in_voltage_sampling_frequency". I do not know the way to change two parameters at runtime. I have only way is add two files "ActiceUseCase.profile" and "ActiveUtilInit.profile" and rebuild the Image kernel to change these params. Do you have any idea to change there param at runtime ? Or any suggestion to modify the driver kernel to add "profile_config" as same adrv9009 ? Or I have to wait the update version in the future? Tks for your answer. ^^

mhennerich, 2024-01-23:
Right now the profile_config is not implemented for ADRV9025, however it should be relatively straight forward to add it in future.

Feel free to create a feature request issue in the github linux repository.

I won't be able to provide a ETA at this point. Alternatively you can try to add it yourself and create a Pull Request.

> Do you have any idea to change there param at runtime ?

Well you can use devicetree overlays.

Applying an overlay should work once.

Removing overlays is currently not supported and may also need some development effort.

https://github.com/analogdevicesinc/linux/blob/main/drivers/jesd204/jesd204-core.c#L1147

ducanh1699, 2024-01-26:
I want to create new profile_config on adrv9025, I based on adrv9009 and adrv9001/9002 try to parse Json file "ActiveUtilInit.profile".
First, I try to mimic all the functions in adrv9009: 
                  sysfs_bin_attr_init(&phy->bin);
                  phy->bin.attr.name = "profile_config";
                  phy->bin.attr.mode = S_IWUSR | S_IRUGO;
                  phy->bin.write = adrv9009_profile_bin_write;
                  phy->bin.read = adrv9009_profile_bin_read;
                  phy->bin.size = 8192;
and the process parse Json file like adrv9002: jsmn.h and jsmn.c
profile_config is already in /sys/bus/iio/devices/iio:device2 but when I echo the file "ActiveUtilInit.profile"  > profile_config the program carsh immediately. 
I think the problem in the  adrv9025_profile_bin_write because 
static ssize_t adrv9025_profile_bin_write(struct file *filp, struct kobject *kobj, struct bin_attribute *bin_attr, char *buf, loff_t off, size_t count)
When I echo  "ActiveUtilInit.profile"  > profile_config, I do not know how to the content in the file can convert to char *buf maybe It need pre-process to convert to char* buf? May you help me step by step to add this feature ^^.
 I think profile_config is very necessarry attribute, so could you please update new feature as soon as prossible ^^. I now try my best to add but It quite complicated.