analogdevicesinc/linux #2424: ADF4360 Incorrect Mask Used.

anthonygclark, 2024-02-08:
ADI,

I'm unsure of the stability of this driver in general but I am attempting to use it - I'll report my success here or on the forum as the documentation is sparse. I'm comparing the register outputs with the EVAL kit Windows software provided by ADI. When doing some checks, it seems this mask below should change from:
```c
val = st->regs[ADF4360_CTRL] & ~ADF4360_ADDR_MUXOUT_MSK;
```

to:
```c
val = st->regs[ADF4360_CTRL] & ~ADF4360_ADDR_PWR_DOWN_MSK;
```

https://github.com/analogdevicesinc/linux/blob/master/drivers/iio/frequency/adf4360.c

mhennerich, 2024-02-08:
Confirmed this is indeed a bug. Thanks for reporting.

anthonygclark, 2024-02-08:
Thanks @mhennerich. I appreciate the attention and time!

closed 2024-02-08 by anthonygclark