analogdevicesinc/linux #2428: ad7768 reset

jamesoncollins, 2024-02-13:
The ad7768-1 performs a softreset using `ret = ad7768_spi_reg_write(st, AD7768_REG_SYNC_RESET, 0x3)`.  Is there a reason that the ad7768 doesn't do the same?

I'm aware that gpio-based reset support was recently added, but this requires that the user have that pin wired whereas the SPI option, I assume, always works:

https://github.com/analogdevicesinc/linux/blob/aa9fdb5fdf095cf2c05ce18444f830b8a05bba4e/drivers/iio/adc/ad7768.c#L834C15-L834C38

nunojsa, 2024-02-13:
Hi,

> Is there a reason that the ad7768 doesn't do the same?

I guess whoever did that driver did not included it... But yes, typically what should be done for devices that allow for soft reset is to first look at gpio based reset and if it's not there, fallback to the soft reset.

That can be easily added but I don't have any board to test it... Are you willing to test it? If so, I can easily push a branch with that change and if all goes well, I then open a PR for it...

jamesoncollins, 2024-02-13:
My build system is many revs behind both this repo and and hdl repo.

I'm going to try and do a build right now with the latest and see if it works.  If so I should be able to test this change.  

I'll post back in a bit.