analogdevicesinc/linux #2688: what's the point of the driver if there is no way to connect the board?

tekntrash, 2025-01-09:
At https://github.com/analogdevicesinc/linux/blob/main/drivers/iio/resolver/ad2s1200.c there is a driver for LINUX for the EVAL-AD2S1210SDZ board 

However, this board depends on another ( SDP-B ) to connect to the computer over USB, and this only connects to the computer via USB using a proprietary method and only for WINDOWS

So, the question is what is the point of offering a driver that cannot be used?

tfcollins, 2025-01-10:
J3 and J4 on EVAL-AD2S1210SDZ can be used to connect another micro which can run Linux or you can use a custom board

tekntrash, 2025-01-10:
Thanks, the EVAL-AD2S1210SDZ is attached to a SDP-B board which has a UART connector. It'd be easier to implement if this connector worked

mhennerich, 2025-01-10:
The AD2S1210 device driver docs have some detailed wiring instructions:
https://wiki.analog.com/resources/tools-software/linux-drivers/iio-resolver/ad2s1210
This board has also the SDP connector...

tekntrash, 2025-01-10:
Thanks. As we understand, no UART support was implemented in the SDP-B

tekntrash, 2025-01-11:
> The AD2S1210 device driver docs have some detailed wiring instructions: https://wiki.analog.com/resources/tools-software/linux-drivers/iio-resolver/ad2s1210 This board has also the SDP connector...

Thanks, the instructions involve desoldering a resistor which is the size of a half grain of rice and find another (impossible to know the resistance) to connect another jumper

Clearly this solution is not commercially minded