analogdevicesinc/linux #2826: How to properly modify CMA size configuration?

whale1st, 2025-06-26:
I’m working with an IIO device and need to allocate larger DMA buffers by adjusting the Contiguous Memory Allocator (CMA) size.    I found some information about CMA on a [webpage](https://wiki.analog.com/resources/tools-software/linux-software/libiio_tips_tricks), but I’m unsure how to correctly interpret or apply it.    Could someone clarify the proper way to modify the CMA size?
```
Linux Contiguous Memory Allocator (or CMA)
Allocation of big, physically-contiguous memory blocks
Reserve memory early at boot time
Kconfig menu "Device Drivers" → "Generic Driver Options"→ "Contiguous Memory Allocator"
Kernel command line option cma=size_in_bytes
PlutoSDR default 256M
Depending on your IIO buffer DMA max block size and number of kernel buffer count you may need to CMA size
```
Any guidance would be greatly appreciated!

mhennerich, 2025-06-27:
Not sure why its not shown by default.
In menuconfig type '/' the search window open search for 'CMA_SIZE_MBYTES' type the number accosiated with that property and hit return you should then be able to modify the default value.

[image]

mhennerich, 2025-06-27:
Maybe not enough memory?